cyanfish/naps2 #300: [Enhancment]: Autmatic scan if feeder is not empty

n00ris, 2024-02-04:
**Is your feature request related to a problem? Please describe.**
I own a Plustek ePhoto z300.

https://plustek.com/us/products/film-photo-scanners/ephoto-z300/

This scanner has a feeder, but only one page at a time. You need to add further pages one by one.

The original software does not provide OCR, so I look for an alternative.

The vendor software has the following feature. 
After the software has been started the scanner is initialized.
Now, if you add a page in the feeder, the scan process starts automatically.
Then you can add future pages in the feeder, it will be scanned and added to the stack.

**Describe the solution you'd like**
I thought the same feature could be added to the "Batch Scan" dialog.

A further option could be added.

Multiple scans (scan if page is added to feeder)

Select you profile (with a feeder functionality).
If you select this option, a further dialog may open. 
Now the scanner is initalized and a poll for the feeder tray is active.
While this dialog box is open:

If a page is added in the feeder, it will be scanned.
User repeats this until happy.
The dialog may have a "done" button.
The scanner / polling could be de-initalized.
The scan process is done.

**Additional context**
You already implemented working detection if the feeder is empty. If the feeder is empty and I push the scan button, a message appears which tells me the feeder is empty. If I add a page to the feeder and hit the scan button in the NAPS2 UI, it scans as expected. 
So the code to detect if the feeder has a page is already present in the code base.

n00ris, 2024-02-04:
Alternative:

The following feature may be part of this feature:

https://github.com/cyanfish/naps2/issues/234

But as a batch mode implementation, that does not need interact with the UI between feeding pages.

Display the Automatic Feeder Detection message until a page is added. After scan show it again.

Repeat until a Done or OK button in UI has beend clicked.

Sounds like a variant of the "Multiple Scans (prombt between scans)" option, but next scan is triggered by the feeder, not UI input.

cyanfish, 2024-02-05:
I'm going to merge this into #234 as that's closely related.

closed 2024-02-05 by cyanfish