cyanfish/naps2 #301: [Idea] Move language to settings

Tarek-Hasan, 2024-02-04:
As NASP2 now includes a settings window, move language option to there from main window.

Tarek-Hasan, 2024-02-04:
Off-topic:
For your information, Empria software, the original developer of PDFsharp has released v6.x for .NET 6. It can be found [here](https://github.com/empira/PDFsharp).

cyanfish, 2024-02-05:
I don't want to remove Language from the main toolbar by default, as if you're opening the app in the wrong language it's a lot easier to identify the globe icon than find something in menus. But per #207 I'm planning on letting users hide toolbar buttons they don't want, including the Language button.

I've had a quick look at PDFsharp 6.x. It looks like it has support for better encryption which is nice but not super important to me, and aside from that I don't think it can replace PdfSharpCore (e.g. it doesn't properly support macOS fonts, and it doesn't seem like they accept third-party PRs).

closed 2024-02-05 by cyanfish

Tarek-Hasan, 2024-02-05:
The cog icon is a universal symbol for settings/options. So I don't think finding the language option will be a problem for people who used a digital electronics in the last 30 years.

cyanfish, 2024-02-06:
I've created #306 to track toolbar change suggestions.