cyanfish/naps2 #306: Toolbar changes/customization

cyanfish, 2024-02-06:
Various ideas suggested include:
- Remove the Language button in favour of putting it in the Settings window
- Move the saving buttons to the right of the editing buttons
- Split the toolbar into multiple draggable components
Also see referenced issues.

I'm not sure I'm going to do any of these changes, but keeping this as a tracking issue.

Tarek-Hasan, 2024-02-10:
Hi,
I think it would be nice to use a broom icon for the clear button, instead of cross inside hexagon icon.

muescha, 2024-12-21:
reorder icons should also available.
for example I like to use `ReorderAltInterleave` as an Icon (and keyboard shortcut)

cyanfish, 2026-10-25:
In [8.6.0](https://www.naps2.com/download) there's some limited customization (ability to remove buttons).