cyanfish/naps2 #314: Separate documents by barcode scanned

dienit251, 2024-02-23:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Instead of separating documents by inserting Patch-T page between documents, I'd like to separate them by the barcodes scanned on pages. So a document will start at the page with barcode ( fixed position ) and end at the page before next barcode scanned. 

**Describe alternatives you've considered**

**Additional context**

Lheriss, 2026-09-27:
Hi,
This isn't something NAPS2 does natively today, but if a standalone tool that watches a folder works for your workflow: I built [pdf-dispatch](https://github.com/Lheriss/pdf-dispatch), a self-hosted Docker service that does exactly this — detects a barcode/QR code anywhere on the page (not tied to a fixed Patch-T position) and splits the batch at each occurrence, matching either an exact value or a pattern (e.g. INVOICE*). Works well as a second step after NAPS2: scan the batch, drop the output PDF into pdf-dispatch's watched folder, get one file per document.

Not a replacement for native support in NAPS2 (which would obviously be more convenient), just flagging it in case it unblocks anyone here in the meantime.

IvarSnaaijer, 2026-10-18:
I would like to pick this up and already have something that passes the initial test, but I do not yet agree with the way it works.
Would you mind if I add an item to the SaveSeparator enum? I propose Patch after which you can select a different (set of) types. and extend the current Barcode option to have parameters for type and simple filtering. PatchT will still be working for backwards compatibility.

IvarSnaaijer, 2026-10-18:
I have been working through the code for this, and it requires quite a lot of changes, it is probably the reason it was not yet implemented.
A lot of the changes are small, but sometimes hard to read. This will therefor require a lot of work from the team to approve.  
I'll be adding test documents in the hope of weeding out problems.