cyanfish/naps2 #318: System.Exception: Could not initialize an instance of the type 'CoreGraphics.CGlmage': handle is null. It is possible to ignore this condition by setting ObjCRuntime.Class.ThrowOn|nitFailure to false.

ten-typek, 2024-02-29:
**Describe the bug**
After the scanning, I receive a error message saying: System.Exception: Could not initialize an instance of the type 'CoreGraphics.CGlmage': handle is null. It is possible to ignore this condition by setting ObjCRuntime.Class.ThrowOn|nitFailure to false.
And no documents are shown, even though the scanning is over. 

**To Reproduce**
Steps to reproduce the behavior:
1. Scan the document
2. Scan will begin
3. After end of the scan, error shows up 

**Expected behavior**
No error should show up. 

**Desktop (please complete the following information):**
 - OS: MacOS 14.2.1
 - Version [e.g. 7.1.0]

**Additional context**
Debug log: 

_2024-02-29 20:48:21.0836 7074 ICC: Opening session 
2024-02-29 20:48:22.9596 7074 DidOpenSession (null) 
2024-02-29 20:48:22.9632 7074 ICC: Waiting for ready 
2024-02-29 20:48:22.9673 7074 DidBecomeReady 
2024-02-29 20:48:22.9673 7074 ICC: Selecting unit 
2024-02-29 20:48:22.9875 7074 DidSelectFunctionalUnit ICScannerFunctionalUnitFlatbed (null) 
2024-02-29 20:48:22.9925 7074 DidSelectFunctionalUnit ICScannerFunctionalUnitFlatbed (null) 
2024-02-29 20:48:22.9925 7074 ICC: Setting scan parameters 
2024-02-29 20:48:22.9968 7074 ICC: Requesting scan 
2024-02-29 20:48:23.1881 7074 DidReceiveStatusInformation ICScannerStatusWarmingUp 
2024-02-29 20:48:24.8263 7074 DidReceiveStatusInformation ICScannerStatusWarmUpDone 
2024-02-29 20:48:37.9275 7074 DidScanToBandData buffer complete 
2024-02-29 20:48:37.9275 7074 DidCompleteScan (null) 
2024-02-29 20:48:37.9283 7074 ICC: Waiting for scan results 
2024-02-29 20:48:37.9283 7074 DidReceiveStatusInformation (null) 
2024-02-29 20:48:37.9283 7074 Flushing image with color sync profile /System/Library/Frameworks/ICADevices.framework/Versions/A/Resources/Scanner RGB Profile.icc 
2024-02-29 20:48:37.9307 7074 ICC: Closing session (in finally) 
2024-02-29 20:48:37.9355 6860 System.Exception: Could not initialize an instance of the type 'CoreGraphics.CGImage': handle is null.
It is possible to ignore this condition by setting ObjCRuntime.Class.ThrowOnInitFailure to false. System.Exception: System.Exception: Could not initialize an instance of the type 'CoreGraphics.CGImage': handle is null.
It is possible to ignore this condition by setting ObjCRuntime.Class.ThrowOnInitFailure to false.
   at ObjCRuntime.DisposableObject.InitializeHandle(NativeHandle handle, Boolean verify)
   at ObjCRuntime.DisposableObject..ctor(NativeHandle handle, Boolean owns, Boolean verify)
   at CoreFoundation.NativeObject..ctor(NativeHandle handle, Boolean owns, Boolean verify)
   at CoreFoundation.NativeObject..ctor(NativeHandle handle, Boolean owns)
   at CoreGraphics.CGImage..ctor(Int32 width, Int32 height, Int32 bitsPerComponent, Int32 bitsPerPixel, Int32 bytesPerRow, CGColorSpace colorSpace, CGBitmapFlags bitmapFlags, CGDataProvider provider, NFloat[] decode, Boolean shouldInterpolate, CGColorRenderingIntent intent)
   at NAPS2.Scan.Internal.Apple.DeviceOperator.FlushImageWithColorSpace(MemoryStream fullBuffer, ICScannerBandData data)
   at NAPS2.Scan.Internal.Apple.DeviceOperator.<>c__DisplayClass26_0.<DidScanToBandData>b__0()
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at NAPS2.Scan.Internal.Apple.DeviceOperator.Scan()
   at NAPS2.Scan.Internal.Apple.AppleScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action`1 callback)
   at NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action`2 callback)
   at NAPS2.Remoting.Worker.WorkerServiceImpl.Scan(ScanRequest request, IServerStreamWriter`1 responseStream, ServerCallContext context)
   at void NAPS2.Remoting.RemotingHelper.HandleErrors(Error error)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.Scan(ScanningContext scanningContext, ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, string> imageCallback)
   at async Task NAPS2.Scan.Internal.WorkerScanBridge.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }_

cyanfish, 2024-02-29:
What version of NAPS2 are you using? There was a [similar issue](https://github.com/cyanfish/naps2/issues/205) that was fixed in 7.1.2.

ten-typek, 2024-03-08:
Hi, I updated the app and the issue is gone, I was using an older version. Thanks for your help and an amazing tool! :)