cyanfish/naps2 #319: PdfSharpCore assembly 1.0.0.0 not found

markus-codechefs, 2024-03-03:
**Describe the bug**
I created a console app .NET 6 which has the NAPS2.SDK nuget installed. I tried out one of the samples with Twain. 

Scanning works fine but upon exporting the result via the following lines of code:

` // Scan and save PDF
 var images = await controller.Scan(options).ToListAsync();
 var pdfExporter = new PdfExporter(scanningContext);
 await pdfExporter.Export("doc.pdf", images);`

I get the following exception:

`Unhandled exception. System.IO.FileNotFoundException: Could not load file or assembly 'PdfSharpCore, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.
File name: 'PdfSharpCore, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null'
   at NAPS2.Pdf.PdfExporter.<>c__DisplayClass5_0.<<Export>b__0>d.MoveNext()
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at NAPS2.Pdf.PdfExporter.<>c__DisplayClass5_0.<Export>b__0()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at NAPS2.Pdf.PdfExporter.Export(OutputPathOrStream output, ICollection`1 images, PdfExportParams exportParams, OcrParams ocrParams, ProgressHandler progress)
   at VersionCustodian.ScanningLayer.Scanner.Scan() in ...\Scanner.cs:line 41
   at Program.Main(String[] args) in ...Program.cs:line 8
   at Program.<Main>(String[] args)`

**Expected behavior**
I see that the NAPS2.SDK csproj has a reference to such a dll via the NAPS2.Setup directly. But the Setup is not in the nuget package. I would expect the package to find the dll.

**Failed Workarounds**
1. Installing the same version via the PdfSharpCore nuget 1.0.0 does not work. For some reason the IImageSource interface is under a different namespace in those versions.

**Fix**
Directly downloading the dll and adding it directly to my project resolved my issue. Nonetheless, I think this should be mentioned in the docs if this is the correct setup. If not some help would be appreciated. :)  

Btw awesome library! I hope to be able to use it further. 

**Desktop (please complete the following information):**
 - OS: Windows 11

cyanfish, 2024-03-03:
Hi, can you try with the latest SDK version (0.5.0)?

Long-term there might be some possibility to upstream my [changes](https://github.com/ststeiger/PdfSharpCore/compare/master...cyanfish:naps2-pdfsharp:master) to PdfSharpCore, but currently it needs to be a separate incompatible DLL.

markus-codechefs, 2024-03-04:
Ah now it's clear your dll comes from a fork from their project which is why their versions would not work. 

What feature of net462 do you require? Is there no alternative in the net standard 2.0?

markus-codechefs, 2024-03-04:
I'll try your new version later

markus-codechefs, 2024-03-04:
Version 0.5.0 has the same issue. I removed the dll in my project and tried to export the image. Workaround still works

cyanfish, 2024-03-07:
Are you sure? In my test it resolved the problem. What target framework are you using? Are you using NAPS2.Sdk directly from your application or through an indirect reference?

markus-codechefs, 2024-03-07:
I'm using net8 projects with the SDK directly as a nugget. 
I'm everything the twain sample code. Ah but I do have 2 projects so technically the SDK is indirect over the second.

cyanfish, 2024-03-07:
Ok, can you try adding a reference to NAPS2.Sdk in proj1?

markus-codechefs, 2024-03-07:
I'll try yes.

markus-codechefs, 2024-03-10:
Alright adding 0.5.0 sdk to the main project and removing the directly added dll solved the issue. Thanks. 

Nit-pick: having to have the naps2.sdk reference in a project where it is not used is a not attractive.

closed 2024-03-10 by markus-codechefs

cyanfish, 2024-03-11:
This should be properly fixed in 0.6.0 (so indirect references work).