cyanfish/naps2 #321: TWAIN: gRPC error: Failed to connect to all addresses

ostasevych, 2024-03-06:
Hi! One of my users updated windows to v.11 and now started to receive errors using Twain connection: it is even not possible to select twain implementation (I use sanetwain):

[image]

[image]

See attached error and debug log.

Naps2 v.7.3.1.0
Win11 x64
Sanetwain
Canon MF3010
[debuglog.txt](https://github.com/cyanfish/naps2/files/14514164/debuglog.txt)
[errorlog.txt](https://github.com/cyanfish/naps2/files/14514165/errorlog.txt)

cyanfish, 2024-03-07:
That indicates the TWAIN process crashed. You can try checking the Windows [Event Viewer](https://www.howtogeek.com/123646/htg-explains-what-the-windows-event-viewer-is-and-how-you-can-use-it/) under Application and see if it has any relevant errors.

ostasevych, 2024-03-10:
So, I've decided to install test sane device on my Win10 laptop via wsl2 and got the same error with default TWAIN settings.
```
Grpc.Core.RpcException: Status(StatusCode="Unavailable", Detail="failed to connect to all addresses")
   at async Task<bool> GrpcDotNetNamedPipes.Internal.MessageReader<TMessage>.MoveNext(CancellationToken cancellationToken)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.TwainScan(ScanOptions options, CancellationToken cancelToken, ITwainEvents twainEvents)
   at async Task NAPS2.Scan.Internal.Twain.RemoteTwainSessionController.StartScan(ScanOptions options, ITwainEvents twainEvents, CancellationToken cancelToken)
   at async Task NAPS2.Scan.Internal.Twain.TwainScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback)+(?) => { }
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async Task NAPS2.Scan.ScanController+<>c__DisplayClass13_1.<Scan>g__DoScan|6(?)+DoScan(?)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
```

The Event manager says:
in `Windows Logs\Application`:

**Error**
in `General`:
```
Назва аварійного застосунку: NAPS2.Worker.exe, версія: 7.3.1.0, позначка часу: 0xdbbed494
Назва аварійного модуля: ntdll.dll, версія: 10.0.19041.3996, позначка часу: 0x9b4c0fa6
Код винятку: 0xc0000374
Зсув помилки: 0x000e6d13
Код аварійного процесу: 0x1ce4
Час запуску аварійного застосунку: 0x01da72ea903eb7e4
Шлях до аварійного застосунку: C:\Program Files\NAPS2\lib\NAPS2.Worker.exe
Шлях до аварійного модуля: C:\WINDOWS\SYSTEM32\ntdll.dll
Код звіту: 2b77b04a-7c2f-40c1-b4ae-a55a211a835e
Повне ім'я аварійного пакету: 
Код аварійного застосунку, що стосується пакету: 
```

Details:
```
- System 
  - Provider 
   [ Name]  Application Error 
  - EventID 1000 
   [ Qualifiers]  0 
   Version 0 
   Level 2 
   Task 100 
   Opcode 0 
   Keywords 0x80000000000000 
  - TimeCreated 
   [ SystemTime]  2024-03-10T12:59:11.5831387Z 
    EventRecordID 1584940 
    Correlation 
   - Execution 
   [ ProcessID]  0 
   [ ThreadID]  0 
    Channel Application 
    Computer Laptop
    Security 
 
- EventData 
   NAPS2.Worker.exe 
   7.3.1.0 
   dbbed494 
   ntdll.dll 
   10.0.19041.3996 
   9b4c0fa6 
   c0000374 
   000e6d13 
   1ce4 
   01da72ea903eb7e4 
   C:\Program Files\NAPS2\lib\NAPS2.Worker.exe 
   C:\WINDOWS\SYSTEM32\ntdll.dll 
   2b77b04a-7c2f-40c1-b4ae-a55a211a835e 
```

**Information:**

in `General`:
Сегмент пам’яті з помилкою: 1724420631852240939, тип 1
Ім’я події: APPCRASH
Відповідь: Недоступно
Код CAB-файлу: 0

Підпис проблеми:
P1: NAPS2.Worker.exe
P2: 7.3.1.0
P3: dbbed494
P4: StackHash_2917
P5: 10.0.19041.3996
P6: 9b4c0fa6
P7: c0000374
P8: PCH_F5_FROM_ntdll+0x0007314C
P9: 
P10: 

Вкладені файли:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERC218.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERC853.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERC892.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERC8A4.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERC903.tmp.txt

Можливе розташування файлів:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_NAPS2.Worker.exe_b010e4dea48e5d722c4adc783a15de47edeb5fb_6a889dc6_a9402d5a-7232-4968-b065-7fa1b382040c

Символ аналізу: 
Повторний пошук рішення: 0
Код звіту: 2b77b04a-7c2f-40c1-b4ae-a55a211a835e
Стан звіту: 268435456
Гешований сегмент пам’яті: 635a33c33279790297ee5f6da4e8502b
GUID CAB-файлу: 0
```

in `Details`:
```
- System 
  - Provider 
   [ Name]  Windows Error Reporting 
  - EventID 1001 
   [ Qualifiers]  0 
   Version 0 
   Level 4 
   Task 0 
   Opcode 0 
   Keywords 0x80000000000000 
  - TimeCreated 
   [ SystemTime]  2024-03-10T12:59:15.7443495Z 
   EventRecordID 1584941 
   Correlation 
  - Execution 
   [ ProcessID]  0 
   [ ThreadID]  0 
   Channel Application 
   Computer Laptop
   Security 

- EventData 
   1724420631852240939 
   1 
   APPCRASH 
   Недоступно 
   0 
   NAPS2.Worker.exe 
   7.3.1.0 
   dbbed494 
   StackHash_2917 
   10.0.19041.3996 
   9b4c0fa6 
   c0000374 
   PCH_F5_FROM_ntdll+0x0007314C 
    
    
   \\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERC218.tmp.dmp \\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERC853.tmp.WERInternalMetadata.xml \\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERC892.tmp.xml \\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERC8A4.tmp.csv \\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERC903.tmp.txt 
   \\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_NAPS2.Worker.exe_b010e4dea48e5d722c4adc783a15de47edeb5fb_6a889dc6_a9402d5a-7232-4968-b065-7fa1b382040c 
    
   0 
   2b77b04a-7c2f-40c1-b4ae-a55a211a835e 
   268435456 
   635a33c33279790297ee5f6da4e8502b 
   0 
```
![image](https://github.com/cyanfish/naps2/assets/5199030/b70fcde5-ba07-41c7-bbdd-eb78e4f6d3d7)

closed 2024-03-11 by cyanfish