cyanfish/naps2 #324: Reduce colour depth option and compression option

NextTherapist, 2024-03-09:
**Is your feature request related to a problem? Please describe.**
I would like to be able to reduce colour depth in NAPS2 after scanning or importing a colour or greyscale image or PDF, before generating a PDF file.

**Describe the solution you'd like**
An option to choose the target colour depth and a preview to judge if these settings lead to a readable result. And then deciding which compression should be used, JPG or CCITT.

Best case would be a new option to use JBIG2 compression optionally.

cubed-it, 2024-04-28:
I have a large number of color scans of documents that should actually be saved in black and white. I'm missing a function like this, especially via the cli.

cyanfish, 2026-10-12:
Click the "Compress" button in the Image menu in [8.5.0](https://www.naps2.com/download).

cyanfish, 2026-10-12:
Also JBIG2 is used by default as well in 8.5.0 for black & white bit depth

closed 2026-10-12 by cyanfish