cyanfish/naps2 #327: SaneWinDS reports System.Exception: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.

ostasevych, 2024-03-10:
I am trying to find out a reliable twain client for windows, sanetwain doesn't want to work with native scanning, and needs to call scanimage every time, wia clears the settings after each restart of the client, and needs to change the id of device every time the client is restarted. Finally I was trying to conquer SaneWinDS which has the most recent developments, and here I have also failed: with the native or memory modes I am constantly getting the error:

```
System.Exception: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.

Server stack trace: 
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   at System.Reflection.RuntimeMethodInfo.UnsafeInvokeInternal(Object obj, Object[] parameters, Object[] arguments)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at DS_EntryWrapper.Message_To_DS(DS_EntryWrapper* , TW_IDENTITY* _pOrigin, UInt32 _DG, UInt32 _DAT, UInt16 _MSG, Void* _pData)
   at NTwain.Triplets.Dsm.NativeMethods.DsmWinNew(TWIdentity origin, TWIdentity destination, DataGroups dg, DataArgumentType dat, Message msg, TWUserInterface data)
   at NTwain.Triplets.Dsm.DsmEntry(TWIdentity origin, TWIdentity destination, Message msg, TWUserInterface data)
   at NTwain.Triplets.UserInterface.EnableDS(TWUserInterface userInterface)
   at NTwain.TwainSession.<>c__DisplayClass128_0.<NTwain.Internals.ITwainSessionInternal.EnableSource>b__0()
   at NTwain.MessageLoopHook.<>c__DisplayClass11_0.<Invoke>b__0(Object o)

Exception rethrown at [0]: 
   at NTwain.MessageLoopHook.Rethrow(Exception ex)
   at NTwain.MessageLoopHook.Invoke(Action action)
   at NTwain.TwainSession.NTwain.Internals.ITwainSessionInternal.EnableSource(SourceEnableMode mode, Boolean modal, IntPtr windowHandle)
   at NTwain.DataSource.Enable(SourceEnableMode mode, Boolean modal, IntPtr windowHandle)
   at NAPS2.Scan.Internal.Twain.TwainSessionScanRunner.Init()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NAPS2.Scan.Internal.Twain.LocalTwainSessionController.<InternalScan>d__8.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NAPS2.Scan.Internal.Twain.LocalTwainSessionController.<StartScan>d__7.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NAPS2.Remoting.Worker.WorkerServiceImpl.<TwainScan>d__21.MoveNext()
   at void NAPS2.Remoting.RemotingHelper.HandleErrors(Error error)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.TwainScan(ScanOptions options, CancellationToken cancelToken, ITwainEvents twainEvents)
   at async Task NAPS2.Scan.Internal.Twain.RemoteTwainSessionController.StartScan(ScanOptions options, ITwainEvents twainEvents, CancellationToken cancelToken)
   at async Task NAPS2.Scan.Internal.Twain.TwainScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback)+(?) => { }
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async Task NAPS2.Scan.ScanController+<>c__DisplayClass13_1.<Scan>g__DoScan|6(?)+DoScan(?)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
```
Though, SaneWinDS works fine as a standalone app.
Tested with a test scanner sane service run in wsl2 on the same PC, as the client.

cyanfish, 2024-03-11:
This is again an issue with the data source, but fortunately SANEWinDS (unlike SaneTwain) is open source, so I was able to make some [fixes](https://github.com/cyanfish/SANEWinDS/pull/1). Hopefully the author is able to merge them, but in the meantime if you like you can trying taking this SANEWinDS.dll and replacing it in the `C:\Windows\twain_32\SANEWinDS` folder.

[SANEWinDS.zip](https://github.com/cyanfish/naps2/files/14552384/SANEWinDS.zip)

ostasevych, 2024-03-11:
> This is again an issue with the data source, but fortunately SANEWinDS (unlike SaneTwain) is open source, so I was able to make some [fixes](https://github.com/cyanfish/SANEWinDS/pull/1). Hopefully [...]

Thanks! I've tested and see that nothing changed significantly. Should I replace the file in the folder "C:\Program Files(x86)\SaneWinDS" as well?

cyanfish, 2024-03-11:
Oops sorry I think I uploaded the wrong build, please try this one instead:
[SANEWinDS.zip](https://github.com/cyanfish/naps2/files/14554316/SANEWinDS.zip)

You might need to right click -> Properties on the DLL and select "Unblock" to get Windows to allow it to run.

You can also update to NAPS2 7.4.0 as it will provide a bit more useful error messages.

ostasevych, 2024-03-11:
> Oops sorry I think I uploaded the wrong build, please try this one instead: [SANEWinDS.zip](https://github.com/cyanfish/naps2/files/14554316/SANEWinDS.zip)
>
> You might need to right click -> [...]

So, what I've done:
1. upgraded portable version to 7.4.0
2. replaced SANEWinDS.dll in c:\windows\twaint_32 and c:\windows\sanewinds folders with the attached one v.1.1.0 (in properties)
3. created profile test with the sanewinds driver

TESTS:

1. test with `default` mode:
Exception UI:
```
Grpc.Core.RpcException: Status(StatusCode="Unknown", Detail="Exception was thrown by handler.")
   at async Task<bool> GrpcDotNetNamedPipes.Internal.MessageReader<TMessage>.MoveNext(CancellationToken cancellationToken)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.TwainScan(ScanOptions options, CancellationToken cancelToken, ITwainEvents twainEvents)
   at async Task NAPS2.Scan.Internal.Twain.RemoteTwainSessionController.StartScan(ScanOptions options, ITwainEvents twainEvents, CancellationToken cancelToken)
   at async Task NAPS2.Scan.Internal.Twain.TwainScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback)+(?) => { }
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async Task NAPS2.Scan.ScanController+<>c__DisplayClass13_1.<Scan>g__DoScan|6(?)+DoScan(?)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
```
in `debug.log`:
```
2024-03-11 14:33:42.1563 1520 Debug: Using new dsm. 
2024-03-11 14:33:42.2888 1520 Using TWAIN DSM: D:\Users\Ostasevych\Downloads\software\naps2-7.3.1-testtwain\App\lib\_win32\twaindsm.dll 
2024-03-11 14:33:42.3049 1520 NAPS2.TW - Opening session 
2024-03-11 14:33:42.3137 1520 Debug: Thread 1: OpenManager. 
2024-03-11 14:33:42.5040 1520 NAPS2.TW - Finishing with error System.TypeInitializationException: The type initializer for 'VBDLL' threw an exception. ---> System.IO.FileLoadException: Could not load file or assembly 'file:///C:\WINDOWS\twain_32\SANEWinDS\SANEWinDS.dll' or one of its dependencies. Operation is not supported. (Exception from HRESULT: 0x80131515) ---> System.NotSupportedException: An attempt was made to load an assembly from a network location which would have caused the assembly to be sandboxed in previous versions of the .NET Framework. This release of the .NET Framework does not enable CAS policy by default, so this load may be dangerous. If this load is not intended to sandbox the assembly, please enable the loadFromRemoteSources switch. See http://go.microsoft.com/fwlink/?LinkId=155569 for more information.
   --- End of inner exception stack trace ---
   at RuntimeAssembly System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, string codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, ref StackCrawlMark stackMark, IntPtr pPrivHostBinder, bool throwOnFileNotFound, bool forIntrospection, bool suppressSecurityChecks)
   at RuntimeAssembly System.Reflection.RuntimeAssembly.nLoad(AssemblyName fileName, string codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, ref StackCrawlMark stackMark, IntPtr pPrivHostBinder, bool throwOnFileNotFound, bool forIntrospection, bool suppressSecurityChecks)
   at RuntimeAssembly System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, ref StackCrawlMark stackMark, IntPtr pPrivHostBinder, bool throwOnFileNotFound, bool forIntrospection, bool suppressSecurityChecks)
   at RuntimeAssembly System.Reflection.RuntimeAssembly.InternalLoadFrom(string assemblyFile, Evidence securityEvidence, byte[] hashValue, AssemblyHashAlgorithm hashAlgorithm, bool forIntrospection, bool suppressSecurityChecks, ref StackCrawlMark stackMark)
   at Assembly System.Reflection.Assembly.LoadFrom(string assemblyFile)
   at static VBDLL()
   --- End of inner exception stack trace ---

Server stack trace: 
   at VBDLL.GetWrapper()
   at DS_EntryWrapper.Log(DS_EntryWrapper* , basic_string<char\,std::char_traits<char>\,std::allocator<char> >* message)
   at NTwain.Triplets.Dsm.NativeMethods.DsmWinNew(TWIdentity origin, TWIdentity destination, DataGroups dg, DataArgumentType dat, Message msg, IntPtr& data)
   at NTwain.Triplets.Dsm.DsmEntry(TWIdentity origin, TWIdentity destination, DataGroups dg, DataArgumentType dat, Message msg, IntPtr& data)
   at NTwain.Triplets.Parent.OpenDsm(IntPtr handle)
   at NTwain.TwainSession.<>c__DisplayClass51_0.<Open>b__0()
   at NTwain.MessageLoopHook.<>c__DisplayClass11_0.<Invoke>b__0(Object o)

Exception rethrown at [0]: 
   at void NTwain.MessageLoopHook.Rethrow(Exception ex)
   at void NTwain.MessageLoopHook.Invoke(Action action)
   at ReturnCode NTwain.TwainSession.Open(MessageLoopHook messageLoopHook)
   at void NAPS2.Scan.Internal.Twain.TwainSessionScanRunner.Init()
2024-03-11 14:33:42.5466 9176 Status(StatusCode="Unknown", Detail="Exception was thrown by handler.") Grpc.Core.RpcException: Status(StatusCode="Unknown", Detail="Exception was thrown by handler.")
   at async Task<bool> GrpcDotNetNamedPipes.Internal.MessageReader<TMessage>.MoveNext(CancellationToken cancellationToken)
[... cut ...]
   at ReturnCode NTwain.Triplets.Dsm.DsmEntry(TWIdentity origin, Message msg, TWIdentity data)
   at ReturnCode NTwain.Triplets.Identity.GetFirst(out TWIdentity source)
   at IEnumerator<DataSource> NTwain.TwainSession.GetEnumerator()+MoveNext()
   at TSource System.Linq.Enumerable.FirstOrDefault<TSource>(IEnumerable<TSource> source, Func<TSource, bool> predicate)
   at void NAPS2.Scan.Internal.Twain.TwainSessionScanRunner.Init()
2024-03-11 14:30:11.2013 24240 Debug: Thread 1: CloseManager. 
```

ostasevych, 2024-03-11:
> Oops sorry I think I uploaded the wrong build, please try this one instead: [SANEWinDS.zip](https://github.com/cyanfish/naps2/files/14554316/SANEWinDS.zip)
>
> You might need to right click -> [...]

So, what I've done:
1. upgraded portable version to 7.4.0
2. replaced SANEWinDS.dll in c:\windows\twaint_32 and c:\windows\sanewinds folders with the attached one v.1.1.0 (in properties)
3. created profile test with the sanewinds driver

TESTS:

1. test with `default` mode:
Exception UI:
```
System.ArgumentException: Unsupported pixel type: 48 2 3 16,16,16,0,0,0,0,0
   at void NAPS2.Scan.Internal.Twain.TwainMemoryBufferReader.CopyBufferToImage(TwainMemoryBuffer memoryBuffer, TwainImageData imageData, IMemoryImage outputImage)
   at void NAPS2.Scan.Internal.Twain.TwainImageProcessor.MemoryBufferTransferred(TwainMemoryBuffer memoryBuffer)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.TwainScan(ScanOptions options, CancellationToken cancelToken, ITwainEvents twainEvents)
   at async Task NAPS2.Scan.Internal.Twain.RemoteTwainSessionController.StartScan(ScanOptions options, ITwainEvents twainEvents, CancellationToken cancelToken)
   at async Task NAPS2.Scan.Internal.Twain.TwainScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback)+(?) => { }
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async Task NAPS2.Scan.ScanController+<>c__DisplayClass13_1.<Scan>g__DoScan|6(?)+DoScan(?)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
```
in `debug.log`:
```
2024-03-11 14:41:16.4351 11028 Debug: Using new dsm. 
2024-03-11 14:41:16.6112 11028 Using TWAIN DSM: D:\Users\Ostasevych\Downloads\software\naps2-testtwain\App\lib\_win32\twaindsm.dll 
2024-03-11 14:41:16.6229 11028 NAPS2.TW - Opening session 
2024-03-11 14:41:16.6401 11028 Debug: Thread 1: OpenManager. 
2024-03-11 14:41:18.7163 20676 Unsupported pixel type: 48 2 3 16,16,16,0,0,0,0,0 System.ArgumentException: Unsupported pixel type: 48 2 3 16,16,16,0,0,0,0,0
   at void NAPS2.Scan.Internal.Twain.TwainMemoryBufferReader.CopyBufferToImage(TwainMemoryBuffer memoryBuffer, TwainImageData imageData, IMemoryImage outputImage)
   at void NAPS2.Scan.Internal.Twain.TwainImageProcessor.MemoryBufferTransferred(TwainMemoryBuffer memoryBuffer)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.TwainScan(ScanOptions options, CancellationToken cancelToken, ITwainEvents twainEvents)
   at async Task NAPS2.Scan.Internal.Twain.RemoteTwainSessionController.StartScan(ScanOptions options, ITwainEvents twainEvents, CancellationToken cancelToken)
   at async Task NAPS2.Scan.Internal.Twain.TwainScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback)+(?) => { }
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async Task NAPS2.Scan.ScanController+<>c__DisplayClass13_1.<Scan>g__DoScan|6(?)+DoScan(?)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
2024-03-11 14:41:25.0825 20676 Killing unresponsive worker 
```

2. test with the `memory` mode:
Exception in UI:
```
System.ArgumentException: Unsupported pixel type: 48 2 3 16,16,16,0,0,0,0,0
   at void NAPS2.Scan.Internal.Twain.TwainMemoryBufferReader.CopyBufferToImage(TwainMemoryBuffer memoryBuffer, TwainImageData imageData, IMemoryImage outputImage)
   at void NAPS2.Scan.Internal.Twain.TwainImageProcessor.MemoryBufferTransferred(TwainMemoryBuffer memoryBuffer)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.TwainScan(ScanOptions options, CancellationToken cancelToken, ITwainEvents twainEvents)
   at async Task NAPS2.Scan.Internal.Twain.RemoteTwainSessionController.StartScan(ScanOptions options, ITwainEvents twainEvents, CancellationToken cancelToken)
   at async Task NAPS2.Scan.Internal.Twain.TwainScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback)+(?) => { }
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async Task NAPS2.Scan.ScanController+<>c__DisplayClass13_1.<Scan>g__DoScan|6(?)+DoScan(?)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
```
in debug.log:
```
2024-03-11 14:42:18.7181 20676 Killing unresponsive worker 
2024-03-11 14:42:24.5590 10896 Debug: Using new dsm. 
2024-03-11 14:42:24.6447 10896 Using TWAIN DSM: D:\Users\Ostasevych\Downloads\software\naps2-testtwain\App\lib\_win32\twaindsm.dll 
2024-03-11 14:42:24.6568 10896 NAPS2.TW - Opening session 
2024-03-11 14:42:24.6707 10896 Debug: Thread 1: OpenManager. 
2024-03-11 14:42:26.3021 20676 Unsupported pixel type: 48 2 3 16,16,16,0,0,0,0,0 System.ArgumentException: Unsupported pixel type: 48 2 3 16,16,16,0,0,0,0,0
[... cut ...]
in debug.log:
```
2024-03-11 14:44:16.8522 24784 Debug: Using old dsm in windows. not C:\WINDOWS\system32\twaindsm.dll 
2024-03-11 14:44:16.9865 24784 Using TWAIN DSM: C:\WINDOWS\twain_32.dll 
2024-03-11 14:44:16.9949 24784 NAPS2.TW - Opening session 
2024-03-11 14:44:16.9949 24784 Debug: Thread 1: OpenManager. 
2024-03-11 14:44:17.0386 24784 NAPS2.TW - StateChanged (to 3) 
2024-03-11 14:44:17.0386 24784 NAPS2.TW - Finding source 
2024-03-11 14:44:25.3412 20676 NAPS2.TW - Sending cancel event 
```

cyanfish, 2024-03-11:
Nice. Does the real scanner work with Default as the twain impl? The "Unsupported pixel type" error is because it's trying to scan 48-bit color but NAPS2 only supports 24-bit color. I'm not sure if that's a bug in SANEWinDS or in the test backend you were using to default to 48-bit.

closed 2024-03-14 by ostasevych

cyanfish, 2024-03-29:
SANEWinDS version 1.2 has [been published](https://sourceforge.net/projects/sanewinds/files/) with those fixes and more, it seems to be working well now.