cyanfish/naps2 #331: Scanner dialog is opened behind the NAPS2 GUI

akuropka, 2024-03-19:
**How to reproduce**
1. Set up a profile using the scanner dialog instead of predefined settings
2. (In my case: Kyocera M5521cdw with TWAIN)
3. Click on 'Scan'

**Expected behavior**
The scanner dialog window is opened in front of the NAPS2 GUI.

**Desktop (please complete the following information):**
 - OS: Microsoft Windows [Version 10.0.22635.3350]
 - NAPS2: [v7.4.0](https://github.com/cyanfish/naps2/releases/tag/v7.4.0)
 - Architecture: x86_64

**Screenshots**
[image]

**Screencast**
https://github.com/cyanfish/naps2/assets/22570574/178bafe5-588b-4fb7-bad0-24c6692739cb

cyanfish, 2024-03-20:
This is an issue with the Kyocera driver. Some things you can try:
- See if Kyocera has any driver updates you can install
- In your profile settings, under Advanced, try different options for "Twain Implementation"
- In your profile settings, use "predefined settings" instead of "native ui"

akuropka, 2024-03-20:
I don't think that this is a driver issue... another program brings it up front ([screencast](https://github.com/cyanfish/naps2/assets/22570574/a44c990c-b88b-47b9-a6f6-b0d2a2480beb)). I am using the scanner dialog of the TWAIN driver because I want to set up parameters before scanning. This is (I think) not possible with NAPS2 except creating a different profile for any possible combination.

cyanfish, 2024-03-23:
Yeah, I do at some point want to add an easier way to tweak settings on the fly, but that's not currently possible. But this is still a Kyocera issue that unfortunately I can't do anything about.

cyanfish, 2024-03-23:
https://github.com/cyanfish/naps2/issues/337 tracks easier setting changes

closed 2024-03-23 by cyanfish