cyanfish/naps2 #332: Help with Sdk Escl network scanner

bernhard-bernan, 2024-03-19:
Sorry for bothering, but would it be possible to get some help with Escl client ? 
I have implemented twine driver and it works fine, but when it comes to network scanners and ESCL i cant seem to get it to work. Cant understand if im missing nuget package or do i need extra configuration for ScanningContext. Right now im just trying to get the network scanners to show in the list of available devices. My code snippet is as follows:
`

     public class ScannerManager { 

        private readonly ScanController _scanController;

        private readonly ScanController _scanNetworkController;

        public ScannerManager()
        {
            var scanningContext = new ScanningContext(new ImageSharpImageContext());
            scanningContext.SetUpWin32Worker();
            _scanController = new ScanController(scanningContext);

            var networkScanningContext = new ScanningContext(new ImageSharpImageContext());
            _scanNetworkController = new ScanController(networkScanningContext);
        }

        public async Task<List<ScannerInfo>> GetLocalScannerInfoAsync()
        {
            return await GetScannerInfo(Driver.Twain);
        }

        public async Task<List<ScannerInfo>> GetNetworkScannerInfoAsync()
        {
            return await GetScannerInfo(Driver.Escl);
        }
        
        private async Task<List<ScannerInfo>> GetScannerInfo(Driver driver)
        {
            List<ScanDevice> devices;
            List<ScannerInfo> scannerInfoList = new();
            if (driver == Driver.Escl)
            {
                devices = await _scanNetworkController.GetDevices(driver).ToListAsync();
            }
            else
            {
                devices = await _scanController.GetDevices(driver).ToListAsync();
            }
            

            foreach (var device in devices)
            {
                scannerInfoList.Add(new()
                {
                    Name = device.Name,
                    DeviceId = device.ID
                });
            }

            return scannerInfoList;
        }
`
For now i have separated them in two if else blocks, just for testing purposes.
Also i have following NAPS2 dependencies for version **0.2.1**:
[image]

Would be really grateful if you could give me some info on what am i missing, because when i try with your NAPS2 desktop app it finds the scanner, but when im trying this example it doesnt work.

cyanfish, 2024-03-19:
Can you try with the latest SDK version (0.6.0)?

Also, this doesn't matter, but you don't need a separate ScanController/ScanningContext for each driver, you can use the same for both.

If 0.6.0 doesn't work, try setting up logging:
`scanningContext.Logger = new ConsoleLogger();`

```c#
public class ConsoleLogger : ILogger
{
    public void Log<TState>(LogLevel logLevel, EventId eventId, TState state, Exception exception, Func<TState, Exception, string> formatter)
    {
        Console.WriteLine(state.ToString());
        if (exception != null) Console.WriteLine(exception.ToString());
    }
    public bool IsEnabled(LogLevel logLevel) => true;
    public IDisposable BeginScope<TState>(TState state) where TState : notnull => throw new NotImplementedException();
}
```

bernhard-bernan, 2024-03-19:
Okay thank you for fast response will try it. Also thanks for pointing out about ScanController i separated ScanController just because i thought win32worker maybe affects it.

bernhard-bernan, 2024-03-20:
Thank you for support, yes updating to 0.6.0 fixed it, retrieval works fine now =)