cyanfish/naps2 #340: Is it possible to use this repo to read in scanned documents into Microsoft Access?

APTGithubAcct, 2024-03-27:
I need to read in documents from a TWAIN printer, show the user a preview, allows them to delete pages if needed, and then finally save the document to disk. Can this repo handle this? If so, how would I get start doing so?

cyanfish, 2024-03-27:
What exactly do you want in terms of integration with MS Access? The NAPS2 [desktop app](https://www.naps2.com) can do all of that (read from twain, show user previews, allow deletion, and save to disk).

But if you want something more tailored, have a look at the [SDK docs](https://www.naps2.com/sdk). You would need to build your own UI as it doesn't come with any. And it's certainly possible to [connect to Access](https://learn.microsoft.com/en-us/visualstudio/data-tools/connect-to-data-in-an-access-database-windows-forms?view=vs-2022) from a .NET app.

APTGithubAcct, 2024-03-28:
> What exactly do you want in terms of integration with MS Access? The NAPS2 [desktop app](https://www.naps2.com) can do all of that (read from twain, show user previews, allow deletion, and save to [...]

I want to build an MS Access form where a user can scan a document in the scanner, hit a button in Access to read the scanner input + display the preview immediately in the form, allow the user to delete a page if needed, and save the document to disk as a PDF. I do not want to use the NAPS2 desktop app because this is part of a larger MS Access program and this functionality needs to live inside of it.

Which of the packages would I need? And do I just save the packages wherever my Access database lives or is there more to the installation process?

closed 2024-04-04 by cyanfish