cyanfish/naps2 #342: Unusual lines in profiles.xml file

03302024, 2024-03-30:
In my profiles.xml file, I have these lines which look unusual:

CustomPageSizeName p3:nil="true" xmlns:p3="http://www.w3.org/2001/XMLSchema-instance"
CustomPageSize p3:nil="true" xmlns:p3="http://www.w3.org/2001/XMLSchema-instance"
KeyValueOptions p3:nil="true" xmlns:p3="http://www.w3.org/2001/XMLSchema-instance"

cyanfish, 2024-03-30:
Well, the schema reference is supposed to be at the top of the file, but it means the element isn't present:
https://www.oreilly.com/library/view/xml-in-a/0596007647/re166.html

03302024, 2024-03-31:
> cyanfish added a commit that referenced this issue

Thank you.

cyanfish, 2024-04-08:
This should be fixed in [NAPS2 7.4.1](https://www.naps2.com/download).

03302024, 2024-04-09:
> This should be fixed in NAPS2 7.4.1

Yes, it is fixed.

03302024, 2024-04-09:
Still, why this line is done this way:
    <CustomPageSizeName xsi:nil="true" />
and doesn't look like this:
    <CustomPageSize>true</CustomPageSize>

cyanfish, 2024-04-09:
CustomPageSizeName is a string (e.g. "File Card"), not a boolean. nil (null) means it's not set

03302024, 2024-04-09:
Thank you for your reply. OK, my question is more about the different line structure used for those three lines. How about this:
    `<CustomPageSizeName>nil="true"</CustomPageSizeName>`
or this:
    `<CustomPageSizeName>nil</CustomPageSizeName>`

cyanfish, 2024-04-09:
It uses xsi:nil instead of just nil as that's the XML standard, so XML readers will interpret it correctly. Putting nil inside the element would mean the page size name is "nil", which isn't the case - the name is absent.

03302024, 2024-04-09:
OK, how about this then:
`<CustomPageSizeName>xsi:nil="true"</CustomPageSizeName>`

cyanfish, 2024-04-09:
That would mean the user has a page size called `xsi:nil="true"`, which granted would be weird, but there's nothing stopping that.

03302024, 2024-04-09:
It is your app, so it is your choice, for sure. 

However, using this term 
xsi:nil="true"
supposedly should have taken care of the confusion.

I am just saying using similar line format to the other lines in the same xml file, would be nice.

cyanfish, 2024-04-09:
I'm not sure why it matters to you, but if you're e.g. trying to parse it, any standard XML library should be able to handle nil attributes. I'm always going to follow the standard conventions where reasonable.

03302024, 2024-04-09:
OK, I will just need to consider this line format when I parse it with my homemade script.