cyanfish/naps2 #347: OCR makes B&W PDF files too big

NextTherapist, 2024-04-04:
**Describe the bug**
OCR makes B&W PDF files uncomprehensibly big.

**To Reproduce**
Steps to reproduce the behavior:

1. B&W-scan a DIN A4 sheet single sided with some B&W text. A typical text/letter sheet with some lines of 12pt black text content. Then save it as PDF A-2b, without doing OCR before. You get a wonderful and small file of about 20-30 KB.

2. Take the same scan, activate OCR (german) and save it again as PDF A-2b. Now you get a file of about 70 KB, but the recognized text makes only about 3-4 KB (difference tested with other OCR Software)

So something in these OCRed files seems to be wrong. Perhaps the file is not saved CCITT compressed but in grey after OCR? I cannot control that.

**Expected behavior**
The file made in 2. should have a maximum size of about 34 KB, not 70 KB.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 7.4.0 32 bit

cyanfish, 2024-04-04:
The extra size is from embedding the font used to render the text, which is required by the PDF-A standard.

NextTherapist, 2024-04-05:
I made some tests: 

NAPS2 embeds a font, when the file contains OCR, independent of the PDF version! Files with OCR contain a subset of Times New Roman, files without OCR do not, and also PDF/A-2b files without OCR do not.

It should not be necessary to embed a font just because there is an invisible OCR text layer in the file.

And of course it would not be necessary to embed a font just because the file is PDF/A. As long as the file content is only a raster image, no font is needed to trusty display the content, and so there is no reason to embed one. But as said, NAPS2 does this right: no font in the scanned and OCR-free PDF/A, the font comes from OCR.

cyanfish, 2024-04-05:
Some OCR software uses a "fake" font instead of embedding a real font, but (a) that means the character measurements are off, which can cause alignment issues, and (b) that can cause various compatibility problems.

In theory it could be possible to provide an option to use a fake font like that, but I'm probably not going to do that.

NextTherapist, 2024-04-25:
I did a test with the "NAPS2 with OCR.pdf" file from above and optimized it with PDF XChange Editor, what mainly means it removed fonts. Result is a file of only 156 KB size, very similar to the PDF24 file.
[NAPS2.with.OCR_Optimized_A2b.pdf](https://github.com/cyanfish/naps2/files/15112594/NAPS2.with.OCR_Optimized_A2b.pdf)

I wanted to try if an embedded font is necessary at all for OCR, but yes, one font is still embedded: It's called "Untitled Truetype (CID) Identity-H" and the precision of OCR positions seems to be fine.

It would be great if NAPS2 could make such a small file by itself.

NextTherapist, 2025-09-29:
Somebody must have made a significant improvement to this issue, for the difference between two ocr'ed files from PDF24 and NAPS2 8.21 only counts 3 KB this morning, without having done anything else in doing OCR. 🥇 

I cannot tell in which version this took place, I updated from 7.53 to 8.21 today. But it is really much better now. ❤️ 
Thank you!

cyanfish, 2026-10-12:
All of these improvements should be in [8.5.0](https://www.naps2.com/download). Glyphless font, JBIG2 compression, more efficient JPEG compression, and other fixes. Let me know if you have any issues.

closed 2026-10-12 by cyanfish

NextTherapist, 2026-11-02:
The improvements lead to wonderfully small pdf files! Thank you!
I personally like JBIG2 compression, but: some archiving experts do not like it since there was a bug in some implementations of the lossy version of JBIG2 (see https://www.dkriesel.com/blog/2015/0317_bsi_verbietet_jbig2).

For this reason, german laws ban the whole algorithm for substitutive scanning (even the lossless one, which I suppose you use in NAPS2). 
So it would be great to have a radio button or a selection to use old G4 instead of JBIG2 in cases where substitutive scanning is demanded.

cyanfish, 2026-11-02:
NAPS2 does of course use only the lossless version. I'm aware of the issues with the lossy version, and lossy seems kind of pointless anyway when lossless is already very small by modern standards.

I can add an option to disable though. https://github.com/cyanfish/naps2/issues/952