cyanfish/naps2 #352: FR: Make Ubuntu & Linux Mint downloads available via a PPA

Gitoffthelawn, 2024-04-09:
**Is your feature request related to a problem? Please describe.**
It's not a big deal, but having to check GitHub and then manually download and install the latest version of NAPS2 from GitHub is a little bit of a hassle.

**Describe the solution you'd like**
Make Ubuntu and Linux Mint downloads available via a PPA.

**Describe alternatives you've considered**
Make Linux downloads available via downstream repositories, such as the ones maintained by Debian, Ubuntu, or Linux Mint.

cyanfish, 2024-04-10:
I've added some instructions here, try this:
https://www.naps2.com/linux-scanning#apt

cyanfish, 2024-04-10:
www.naps2.com and downloads.naps2.com are different physical servers (though it's fair to say the separation isn't perfect as they're both under the same VPS account).

I would be happy to use Github but I don't know that it's possible - all the doc I had was based on local packages. And I don't think PPAs will work as Launchpad doesn't support binary packages.

The main security benefit you get is that you only download the signature once - in the event of a server compromise, you're only vulnerable if you're trying to do the initial Apt setup. And that's no less true if you're just downloading binaries directly.

Gitoffthelawn, 2024-04-18:
I followed your instructions, and they worked well.  Thank you.

There has yet to be a NAPS2 update since you posted your reply, so when you release the next version, I'll know for sure if everything is a success.

If you have a few minutes, and feel like it, and want to explain a bit more why you think a PPA won't work for NAPS2, I'm interested in reading your thoughts.  To date, I have not published a PPA myself, so I'm interested in learning more about potential obstacles.  If you're too busy, or just don't feel writing more on the topic, that's not a problem, and I certainly understand.

cyanfish, 2024-04-20:
NAPS2 currently only builds binary packages, not source packages. PPAs are published through Launchpad, which [does not allow](https://help.launchpad.net/Packaging/PPA) binary packages.
> Note: We do not allow uploading pre-built binary packages. 

 And I'm not sure that making source packages would help, as I assume Launchpad can only use build dependencies that are provided through Ubuntu. NAPS2 uses the latest .NET SDK to build, and I usually upgrade to the newest version on release. But the Ubuntu-based releases tend to [lag behind](https://learn.microsoft.com/en-us/dotnet/core/install/linux-ubuntu#supported-distributions) the official Microsoft releases and aren't supported on all the Ubuntu versions I support.

Gitoffthelawn, 2024-04-23:
Thanks.  How are you defining "source package".  The NAPS2 source is open (a big thank you!), and a makefile can be created to automate the build process.  I get the sense there is more to it though.  Is there?

Regarding the latest .NET SDK, yes, I could see that possibly being an issue.  I'm currently not sure of Launchpad's dependency criteria either.

cyanfish, 2024-04-23:
https://wiki.debian.org/Packaging/SourcePackage
https://wiki.debian.org/Packaging/BinaryPackage

kgraefe, 2024-07-15:
> I've added some instructions here, try this: https://www.naps2.com/linux-scanning#apt

Thanks for that. Minor improvement: Apt handles armored key if the file extension is `.asc`:

```sh
# Download the NAPS2 public key
sudo curl -fsSL https://www.naps2.com/naps2-public.pgp -o /etc/apt/keyrings/naps2.asc

# Add NAPS2 as an Apt source
echo "deb [signed-by=/etc/apt/keyrings/naps2.asc] https://downloads.naps2.com ./" | sudo tee /etc/apt/sources.list.d/naps2.list >/dev/null

# Install NAPS2
sudo apt update
sudo apt install naps2
```

timur-g, 2026-10-12:
This would be small but useful fix, to make known repo is actually available. 
I used  curl -fsSL https://www.naps2.com/naps2-public.pgp | gpg --dearmor | sudo tee  /etc/apt/keyrings/naps2.gpg and also .gpg in source.