cyanfish/naps2 #353: OCR text alignment in NAPS 7.4.1

NextTherapist, 2024-04-09:
It should be improved according to the changelog. But this is what I get:

[7.4.1.pdf](https://github.com/cyanfish/naps2/files/14920377/7.4.1.pdf)

[image]

cyanfish, 2024-04-09:
Do you have a copy of the PDF before running OCR? And did you do any editing other than running OCR?

Also, what are your exact OCR settings? Language is German? Fast/Best? Is "Fix white balance" checked?

NextTherapist, 2024-04-10:
> Do you have a copy of the PDF before running OCR? And did you do any editing other than running OCR?
> 
> Also, what are your exact OCR settings? Language is German? Fast/Best? Is "Fix white balance" checked?

Input file was this: 
[B-W without OCR.pdf](https://github.com/cyanfish/naps2/files/14928626/B-W.without.OCR.pdf)

OCR settings are German, fast, no white balance.

cyanfish, 2024-04-11:
Can you try this test version and see if it resolves the issue for you?

[naps2-7.4.1-win-testocr.zip](https://github.com/cyanfish/naps2/files/14941314/naps2-7.4.1-win-testocr.zip)

NextTherapist, 2024-04-11:
No, it's the same thing:
[naps2-test.pdf](https://github.com/cyanfish/naps2/files/14943728/naps2-test.pdf)

I can remember I saw this before in an older version (before 7.4.0). Perhaps it was 7.3.0 or 7.3.1.

cyanfish, 2024-04-11:
That's very weird, as using that test version, and running on the B-W.without.OCR.pdf file you provided, this is what I get:
[naps2-test2.pdf](https://github.com/cyanfish/naps2/files/14950198/naps2-test2.pdf)

NextTherapist, 2024-04-12:
That's really weird. I thought it came from your PDF version (1.4) and tried the same, but it was the same error. I uninstalled 7.4.1 and installed 7.4.0 again and 7.4.0 is without that error:
[naps2-test3-7.4.0.pdf](https://github.com/cyanfish/naps2/files/14956154/naps2-test3-7.4.0.pdf)

But it is some KB bigger than your file. At least on my system (Windows 10 64bit) 7.4.1 works not like 7.4.0 any more.

cyanfish, 2024-04-14:
This should be fixed in [7.4.2](https://www.naps2.com/download). Thanks for the help with testing!

closed 2024-04-14 by cyanfish

NextTherapist, 2024-04-15:
@cyanfish Thank you for fixing it! :-)