cyanfish/naps2 #356: Error when OCR after upgrade to 7.4.1

ostasevych, 2024-04-10:
Hi! After upgrading from 7.4.0 to 7.4.1 I am constantly getting error when performing OCR job:
```
System.FormatException: Input string was not in a correct format.
   at float Number.ParseSingle(string value, NumberStyles options, NumberFormatInfo numfmt)
   at (float m, float b) NAPS2.Ocr.TesseractOcrEngine.GetBaselineParams(XElement element)
   at OcrResult NAPS2.Ocr.TesseractOcrEngine.CreateOcrResult(XDocument hocrDocument)
   at async Task<OcrResult> NAPS2.Ocr.TesseractOcrEngine.ProcessImage(ScanningContext scanningContext, string imagePath, OcrParams ocrParams, CancellationToken cancelToken)
```
Win10 x64
NAPS2 7.4.1

ostasevych, 2024-04-10:
> Do you have a sample (non-ocr'd) PDF you can attach with the problem? What OCR language are you using?

Any file which I try to OCR after scanning or after importing pdf file. I tried Ukrainian and Ukrainian+English.

cyanfish, 2024-04-10:
I can't reproduce the problem, so a sample would be helpful.

zisco67, 2024-04-12:
Same here - used language is German; downgrade solves the problem!

cyanfish, 2024-04-12:
@zisco67 Do you have a sample (non-ocr'd) PDF you can attach with the problem?

I can't help if I don't know what the problem is.

cyanfish, 2024-04-14:
This should be fixed in [7.4.2](https://www.naps2.com/download).

closed 2024-04-14 by cyanfish

ostasevych, 2024-04-14:
Hi! Thanks, it is fixed.