cyanfish/naps2 #360: Add ESCL driver by manual IP adress

amber8706, 2024-04-11:
NAPS now automatically searches for the ESCL source. But, the search occurs in the current subnet. If the ESCL IP address is in a different subnet, NAPS will not find anything.
Please add an "IP Address" text field so that I can manually specify the IP of a device on a different subnet.

[image]

Thanks.

cyanfish, 2024-04-11:
Depending on your network setup there are some things you can try:
https://www.naps2.com/doc/scanner-sharing#subnet-boundaries

amber8706, 2024-04-11:
I would be satisfied with the "IP address" text field, where I could specify the device's IP manually. As practice has shown, the device search mechanism is not reliable. This is not only a NAPS problem. Firewall rules often block certain ports and IP addresses. Although the device is actually available, it may not be detected on the network. 
No one will change the settings of the entire network for the sake of running a single program. For example, in our organization, not all network MFPs can be detected when installing the driver. This is normal, because for security reasons, these devices are not visible. This allows you to hide at least part of the network infrastructure from malicious hackers. Entering an IP address is normal for us. I would like to have this feature in NAPS as well.

ihipop, 2024-06-14:
[image]

related: https://github.com/alexpevzner/sane-airscan/issues/341

cyanfish, 2024-08-11:
There is now a "Manual IP" option when you select the ESCL driver in [7.5.0](https://www.naps2.com/download).

closed 2024-08-11 by cyanfish

amber8706, 2024-08-27:
Cool! Thanks! I was successfully able to connect the ESCL scanner to another sub-network. I left the "port" text field empty because I didn't know the port number. I specified only the IP address, I did not touch the other parameters. Thanks!