cyanfish/naps2 #364: virustotal.com finding

beauw, 2024-04-13:
**Describe the bug**
The site virustotal.com reports two findings for naps2-7.4.1-win.exe.

The findings are:
1.  Bkav Pro W32.AIDetectMalware
2. The sandbox Yomi Hunter identifies the file as malware

The report is available here:
https://www.virustotal.com/gui/file/5077f9d647b4132f6a41fa9d6d5a8166bd854a2beeb75ef8d5db3be97c43424b/detection

The sandbox details on behavior are here:
https://www.virustotal.com/gui/file/5077f9d647b4132f6a41fa9d6d5a8166bd854a2beeb75ef8d5db3be97c43424b/behavior

cyanfish, 2024-04-13:
There's not much I can do about that, it's pretty clearly a false positive (especially given 68/70 say it's fine).

closed 2024-04-13 by cyanfish

isaak654, 2024-04-14:
For a better understanding, I recommend the following documentation page:
https://docs.virustotal.com/docs/false-positive-contacts

isaak654, 2024-04-14:
For a more in-depth understanding, I recommend the following documentation page:
https://docs.virustotal.com/docs/false-positive-contacts