cyanfish/naps2 #366: Scan in background don't work

Moosbueffel, 2024-04-14:
**Describe the bug**
During scanning, the dialog provide a button to scan in background. If I click this button and than try to work with the previous scanned papers in the application than the application will be greyed and no action is possible. The scanning dialog than freezes and the progress bar will not be updated any more.

After scanning is finished, the scan dialog closes and I can work with the application.

**Expected behavior**
I expect, that I can work with the main application: drag&drop, save, delete scanned papers. And the scanning process will work in the background.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 10.0.22621.2506

cyanfish, 2024-04-14:
What version of NAPS2 are you using? What driver type (twain/wia) and what scanner model?

Moosbueffel, 2024-04-14:
The latest version of NAPS2.
TWAIN
Epson Stylus Office BX935FWD via Network (LAN)

Moosbueffel, 2024-04-15:
Latest version mean: 7.4.2.0

zwo1a, 2024-08-29:
I'm having the same issue, the device I use is Konica Minolta bizhub 554e via Network (WiFi), TWAIN.
OS: Windows 10 x64