cyanfish/naps2 #376: How to batch scan pdf-files to specified folders   by sdk

800903, 2024-04-22:
how to set one file per page 
ScanningContext scanningContext = new ScanningContext(new GdiImageContext());
            var scanController = new ScanController(scanningContext);
            var devices = await scanController.GetDeviceList(Driver.Wia);          
            var options = new ScanOptions
            {
                Device = devices.First(),
                PageSize = PageSize.A4,
                PaperSource = NAPS2.Scan.PaperSource.Flatbed,
                Dpi = 600
            };
            var images =await scanController.Scan(options).ToListAsync();
            var pdfExporter = new PdfExporter(scanningContext);
            await pdfExporter.Export("doc.pdf", images); ?????  ???????  specified dynamic  folders

800903, 2024-04-22:
I don't know if it will be effective.

 
 public async Task ScanWithTwain(string filepath)
        {
            
            ScanningContext scanningContext = new ScanningContext(new GdiImageContext());
            var scanController = new ScanController(scanningContext);
            var devices = await scanController.GetDeviceList(Driver.Wia);
            var options = new ScanOptions
            {
                Device = devices.First(),
                PageSize = PageSize.A4,
                PaperSource = NAPS2.Scan.PaperSource.Flatbed,
                Dpi = 600
            };           
            var images =await scanController.Scan(options).ToListAsync();
            var pdfExporter = new PdfExporter(scanningContext);
            string currentFileName = "";
            int pages= 0;
            var compat = PdfCompat.Default;
            PdfSettings pdfSettings = new PdfSettings();
            pdfSettings.Compat = compat;
            pdfSettings.SinglePagePdfs = true;
            pdfSettings.DefaultFileName = "0000.pdf";
            var imagesByFile = pdfSettings.SinglePagePdfs ? images.Select(x => new[] { x }).ToArray(): new[] { images.ToArray() };
            foreach (var imagesForFile in imagesByFile)
            {
                {
                    pages++;
                    currentFileName = filepath + pages.ToString("D4") + ".pdf";
                    await pdfExporter.Export(currentFileName, imagesForFile, new PdfExportParams(pdfSettings.Metadata, pdfSettings.Encryption, pdfSettings.Compat));
                }
            }
            
        }

cyanfish, 2024-04-22:
First, I'd recommend updating to the latest SDK version, where PdfSettings isn't available because it doesn't do anything.

But as far as code, you don't need any special NAPS2 functionality, just a for loop:
```c#
int i = 1;
foreach (var image in images) {
    await pdfExporter.Export($"{i++:D4}.pdf", new[] { image });
}
```

cyanfish, 2024-04-23:
Well, you'll need to put in a full file path depending on where you want to save it.

800903, 2024-04-23:
Here's the full code.
   public async Task ScanWithWia()
        {
            ScanningContext scanningContext = new ScanningContext(new GdiImageContext());
            var scanController = new ScanController(scanningContext);
            var devices = await scanController.GetDeviceList(Driver.Wia);
            var options = new ScanOptions
            {
                Device = devices.First(),
                PageSize = PageSize.A4,
                PaperSource = NAPS2.Scan.PaperSource.Flatbed,
                Dpi = 600
            };
            var images = await scanController.Scan(options).ToListAsync();
            var pdfExporter = new PdfExporter(scanningContext);
            int pages = 0;
            foreach (var image in images)
            {
                pages++;
                await pdfExporter.Export("D:\\" + pages.ToString("D4") + ".pdf", new[] { image });
            }
        }

800903, 2024-04-23:
[image]

800903, 2024-04-23:
[image]

800903, 2024-04-24:
[image]
this is the value of images.  maybe is wrong

800903, 2024-04-26:
ScanningContext scanningContext = new ScanningContext(new GdiImageContext());
 // got it done  must be add  FileStorageManager  object, Otherwise, the pdf file cannot be saved
FileStorageManager fileStorageManager = new FileStorageManager(@"c:\temp");          
scanningContext.FileStorageManager = fileStorageManager;
//=======================================================

closed 2024-04-27 by cyanfish