cyanfish/naps2 #379: windows driver twain Scanning with Duplex and A3 pagesize incorret

hfbi360, 2024-04-26:
**Describe the bug**
Scanning with "Paper source" set to "Duplex" and with "Page size" set to "A3" and twain scan image pagesize incorrect, but with wia is correct.

when I use Naps2.Console.exe --driver twain --device "PaperStream IP fi-7460 #3
" --source "duplex" --pagesize A3 , it scan the  page  like A4 size but use --driver via the page size is correct.

**Expected behavior**
use driver twain scan A3 duplex can output correct image like use driver wia.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 7.4.2

cyanfish, 2024-04-27:
Sounds like a bug in the PaperStream TWAIN driver, there isn't much I can do about that other than suggest looking for updated drivers.

closed 2024-04-27 by cyanfish

hfbi360, 2024-04-28:
but I use the C++ twain lib  it  can scan A3 right with the same PaperStream TWAIN  driver。
It's a bit similar to this question  #281