cyanfish/naps2 #380: use sdk1.0.0 to scan files ,how to show the ShowOperation  Progress

800903, 2024-04-26:
use demo  from the website -->https://www.naps2.com/sdk
the demo can not show the  scan Progress..   vs2013

cyanfish, 2024-04-27:
https://www.naps2.com/sdk/doc/api/NAPS2.Scan.ScanController.html#events

800903, 2024-04-30:
thanks ,get it done;

 delegate void SetProcssBarCallback(int _value);
    private void SetProcssBar(int _value)
    {
        if (progressBarScan.InvokeRequired)
        {
            while (progressBarScan.IsHandleCreated == false)
            {
                if (progressBarScan.Disposing || progressBarScan.IsDisposed) return;
            }
            SetProcssBarCallback d = new SetProcssBarCallback(SetProcssBar);
            progressBarScan.Invoke(d, new object[] { _value });
        }
        else
        {
            progressBarScan.Value = _value;
        }
    }

//==============================
  var op = new ScanOperation(options);
            scanController.PageStart += (sender, args) =>
            {
                op.NextPage(args.PageNumber);
                SetProcssBar(10);
            };
            scanController.ScanEnd += (sender, args) =>
            {
                op.Completed();
                if (args.Error != null)
                {
                    MessageBox.Show(args.Error.ToString());
                }
                SetProcssBar(100);
            };

closed 2024-05-01 by cyanfish