cyanfish/naps2 #383: EMail PDF doesn't work anymore

EMFAdmin, 2024-05-06:
**Describe the bug**
Since NAPS 7 EMail PDF doesn't work anymore. As Mailclient we use Lotus Notes 9.
With Naps 6.1.2 it works without any problems, by clicking on Email PDF a new window opens in Lotus notes and the scanned PDF file is attached to the mail.

With the actual installed version of NAPS (7.4.0.0) an error is shown "An error occured when trying to send the email", with following Technical Details:
Grpc.Core.RpcException: Status(StatusCode="Unavailable", Detail="failed to connect to all addresses")
   at async Task<bool> GrpcDotNetNamedPipes.Internal.MessageReader<TMessage>.MoveNext(CancellationToken cancellationToken)
   at async Task<TMessage> GrpcDotNetNamedPipes.Internal.MessageReader<TMessage>+<>c__DisplayClass9_0.<ReadNextMessage>b__0(?)+(?) => { }
   at async Task<MapiSendMailReturnCode> NAPS2.Remoting.Worker.WorkerServiceAdapter.SendMapiEmail(string clientName, EmailMessage message)
   at async Task<MapiSendMailReturnCode> NAPS2.ImportExport.Email.Mapi.MapiDispatcher.SendEmail(string clientName, EmailMessage message)
   at async Task<bool> NAPS2.ImportExport.Email.Mapi.MapiEmailProvider.SendEmail(EmailMessage message, ProgressHandler progress)+(?) => { }
   at async bool NAPS2.Pdf.SavePdfOperation.Start(string fileName, Placeholders placeholders, ICollection<ProcessedImage> images, PdfSettings pdfSettings, OcrParams ocrParams, EmailMessage emailMessage, string overwriteFile)+(?) => { } [1]

**To Reproduce**
Steps to reproduce the behavior:
1. select scanned file
2. Click on 'Email PDF'
3. See error

**Expected behavior**
The expected behavior is that a new email window opens and the file is attached to it

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 7.4.0.0

**Additional context**

[image]

800903, 2024-05-06:
version 7.4.2.0 is work by test

cyanfish, 2024-05-11:
@EMFAdmin Can you confirm if 7.4.2 has the same issue?

EMFAdmin, 2024-05-13:
Yes, this version has the same issue. I have tested it with IBM Notes 9 and HCL Notes 12 and the problem is the same as in the screenshot above. With NAPS 6 there are no problems

EMFAdmin, 2024-05-15:
Provider is HCL Notes and Notes is set as standard app for mailto

cyanfish, 2024-07-06:
Please try this with [7.4.3](https://www.naps2.com/download).

closed 2024-07-06 by cyanfish