cyanfish/naps2 #387: Is naps2.net a legit website?

najjara, 2024-05-12:
**Describe the bug**
A quick online search for NAPS2 yields 2 websites:
The official one listed in "About" in this repo (`naps2.com`), and another ending in `.net`

The one ending in `.net` links under the "Download" button to version 7.3.1 directly from this repo (for example https://github.com/cyanfish/naps2/releases/download/v7.3.1/naps2-7.3.1-linux-x64.deb).

It has a poorly written and generic Terms and Conditions (for example "Any claim related to NAPS2’s Website shall be governed by the laws of us without regards to its conflict of law provisions." "trade mark law"), and Privacy Policy that allows for the collection of personal information among other things to do with them as they please. They also refer to "the Company". The Privacy Policy (which has `-2` in the link) mentions `Last updated: March 11, 2024`.

On "About Us" there's an email `support@borderlessgaming.net`. However, the rest of the website, lists the email `support@naps2.net`. There's also a US phone number, and a residential address in Oregon.

The domain was created on 2024-02-08, and the owner's information is obviously hidden behind a domain privacy service. I'm not going to try that `@naps.net` address, but `dig` doesn't show any MX records associated with that domain.
The domain `borderlessgaming.net` was created and registered at the same registrar at the same exact time. Similarly, it steals another open source project (https://github.com/Codeusa/Borderless-Gaming), has no MX records associated with it domain, and they both have the same design (same design template?). It should be no surprise that they share the same hosting company (the same IP even), which is also their registrar.

[image]
[image]
[image]

NextTherapist, 2024-05-14:
We do not offer the newest version since we cannot keep up with those permanent updates. 😂

cyanfish, 2024-05-15:
It's definitely not legit but there's not a whole lot I can do about it.

najjara, 2024-05-15:
You can file DMCA requests with the hosting company, or at least all major search engines. That way at least the deceptive site doesn't show in search results.

najjara, 2024-05-15:
DuckDuckGo shows a badge next to some sites indicating that they're official (for example youtube-dl). I have no idea how to get such a badge

cyanfish, 2024-05-15:
Re: DMCA, I did try with Google but they refused for unknown reasons. And there's a very narrow path for that, the only reason it should have been viable was that they lifted some content straight from naps2.com. Otherwise it's a trademark issue which is hard to enforce, especially without a registered trademark.

MG-LSJ, 2024-05-24:
We can report it on safe browsing as a "malicious" site

https://safebrowsing.google.com/safebrowsing/report_general/

cyanfish, 2024-05-24:
It's not distributing malware, so that isn't suitable.

najjara, 2024-05-24:
Have you tried directly with the hosting company?

najjara, 2024-05-24:
It doesn't have to be a trademark issue. It's a copyright issue, which absolutely belongs to you, and is protected under US law (where the site is hosted).

cyanfish, 2024-05-24:
NAPS2 is freely licensed for redistribution. Anyone is allowed by the GPL license to create a new site to distribute NAPS2 if they like. The issues are (1) they are misrepresenting themselves (trademark) and (2) portions of the homepage have been copied (copyright).

closed 2024-07-15 by cyanfish