cyanfish/naps2 #388: NAPS2 for Android

aurbus, 2024-05-16:
**Is your feature request related to a problem? Please describe.**
no it is not. 

**Describe the solution you'd like**
It would be very helpful to have a version of NAPS2 for Android. It would be tremendously helpful to have it on my Tab S9 Ultra. Android tablets in general have gotten much better, and more people are using them for productivity work like NAPS2. I realize this is probably a rather large request, and I very much understand if it isn't possible. I am only referring to an APK which we would be able to download/install, no app stores or anything like that. 

**Additional context**
very loosely related to #357

cyanfish, 2024-05-16:
Most scanners wouldn't have drivers that work with Android. While it would be possible in principle to have a NAPS2 Android app that works only with ESCL (network) drivers, the reality is that's very niche, and maintaining support for 3 operating systems (Windows/Mac/Linux) is already stretching the limits of what I can handle as one person.

cyanfish, 2024-05-16:
I will add that [Mopria Scan](https://play.google.com/store/apps/details?id=org.mopria.scan.application) does exist as an ESCL Android app if you're not aware. While it doesn't have most of NAPS2's functionality it does have basic PDF scanning capabilities.

aurbus, 2024-05-16:
Ah, yes...I understand... I would guess that even if drivers were not an issue and Android version would probably still be rather niche. I will have a look at Mopria Scan. Thanks for the great work.

closed 2024-05-16 by aurbus