cyanfish/naps2 #389: ESCL no longer working with v 7.4.2.0  - Windows 11

TungstenE2, 2024-05-18:
ESCL no longer working with v 7.4.2.0  - Windows 11

Brother Scanner starts, but no pages are received by NAPS2.

Changed to TWAIN driver, this is working. ESCL was working fine before.

Please check

rousseldenis, 2024-05-18:
I can say 7.4.2 is working fine on Win10

ps1noob, 2024-05-18:
> ESCL no longer working with v 7.4.2.0 - Windows 11
> 
> Brother Scanner starts, but no pages are received by NAPS2.
> 
> Changed to TWAIN driver, this is working. ESCL was working fine before.
> 
> Please check

Can confirm, Brother Scanner, NAPS2 7.4.2.0 with ESCL driver on Windows 11, no pages were received by NAPS2 after the scan process. Using TWAIN it works as expected.

cyanfish, 2024-07-06:
Please try this with [7.4.3](https://www.naps2.com/download).

closed 2024-07-06 by cyanfish