cyanfish/naps2 #394: OCR not working

marrajon, 2024-05-22:
**Describe the bug**
Previously this was my favourite OCR software, even I made a donation , but  it's not working now.
OCR not working, it only produce nonsense characters, Language Spanish. Last version (7.4.2.0)
Debug file not created even if tick is enabled inside 'About'
I've desinstalled and reinstalled, and the behaviour is teh same, even with a portable installation (6.1.2.25879)

**Desktop (please complete the following information):**
 - OS: Windows 10 updated 22/5/2024

Add any other context about the problem here.
[adj-adj-CM23GR_Exclusiones_def.pdf](https://github.com/cyanfish/naps2/files/15407681/adj-adj-CM23GR_Exclusiones_def.pdf)

marrajon, 2024-07-19:
After several trials, now I think the attached pdf is very difficult for tesseract based software.

cyanfish, 2024-08-15:
Closing as this seems more an issue with Tesseract than NAPS2.

closed 2024-08-15 by cyanfish