cyanfish/naps2 #396: Scan with other DPI-resolutions than the given ones

frederikhabils, 2024-05-26:
I would love the ability to add custom scan resolutions to NAPS2, suitable to one's printer. 
In my case (Epson XP-3100), there is a resolution of 240dpi that sits in the sweet spot between 200-300 dpi.
I can use that 240dpi by using the "Use Native UI" option, but that's slower than using "Predefined settings".

So please, let us add custom scan resolutions, e.g. via one of the XML config files (or via the UI) !

ps: tested with Portable naps2 (windows) 7.4.2

**Additional context**
This was previously mentioned in the naps2 discussion/tickets on sourcefourge, but never got reaction, perhaps because developers look only at github issues nowadays ?
e.g. https://sourceforge.net/p/naps2/discussion/general/thread/c06dd63be1/

cyanfish, 2024-08-11:
I haven't added UI for this yet but in 7.5.0 if you want you can edit your profiles.xml file and e.g. replace Dpi300 with Dpi240.

frederikhabils, 2024-10-27:
Had some tech issues, but tested this now with 7.5.2 and it works like a charm :) Obviously will be better with GUI support, but for an informed user it's very usable. Thanks !

cyanfish, 2025-03-21:
This should be added in [NAPS2 8.1.0](https://www.naps2.com/download).

closed 2025-03-21 by cyanfish