cyanfish/naps2 #402: Noisy output when scanning grayscale - Brother DCP-L2500D

silviucc, 2024-06-16:
**Describe the bug**

When scanning using grayscale I get very noisy output

**To Reproduce**
Steps to reproduce the behavior:

1. Configure scanner to use grayscale mode
2. Scan document

[naps2_grayscale.pdf](https://github.com/user-attachments/files/15857187/naps2_grayscale.pdf)

**Expected behavior**

It should output a clean scan result

[simplescan_grayscale.pdf](https://github.com/user-attachments/files/15857198/simplescan_grayscale.pdf)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04
 - Version 7.4.2

**Additional context**
Add any other context about the problem here.

It may be a case where the scanner can do more grayscale scanning modes and simplescan is just selecting the one giving better results by default

NextTherapist, 2024-06-17:
What you call a "noisy" output seem to be shades of grey described via error diffusion. Your "clean scan result" is a simple black and white colour reduction.

Somewhere in your scanner driver (not in NAPS2) there must be a setting like error diffusion chosen instead of black and white.

silviucc, 2024-06-17:
Hi.
Below is the output of scanimage. It seems NAPS2 does not show all the modes supported by the driver.

```
scanimage -A
Output format is not set, using pnm as a default.
scanimage: rounded value of br-x from 211.9 to 211.881
scanimage: rounded value of br-y from 355.6 to 355.567

All options specific to device `brother4:bus7;dev3':
    --mode Black & White|Gray[Error Diffusion]|True Gray|24bit Color[Fast] [24bit Color[Fast]]
        Select the scan mode
    --resolution 100|150|200|300|400|600|1200|2400|4800|9600dpi [200]
        Sets the resolution of the scanned image.
    --source FlatBed [FlatBed]
        Selects the scan source (such as a document-feeder).
    --brightness -50..50% (in steps of 1) [inactive]
        Controls the brightness of the acquired image.
    --contrast -50..50% (in steps of 1) [inactive]
        Controls the contrast of the acquired image.
    -l 0..211.9mm (in steps of 0.0999908) [0]
        Top-left x position of scan area.
    -t 0..355.6mm (in steps of 0.0999908) [0]
        Top-left y position of scan area.
    -x 0..211.9mm (in steps of 0.0999908) [211.881]
        Width of scan-area.
    -y 0..355.6mm (in steps of 0.0999908) [355.567]
        Height of scan-area.

```

NextTherapist, 2024-07-03:
As I said:  the setting "Bit depth: Grayscale" is not what you want. What other options are available for "Bit depth"?

cyanfish, 2024-08-11:
This should be fixed in [7.5.0](https://www.naps2.com/download).

closed 2024-08-11 by cyanfish