cyanfish/naps2 #409: No way to delete a scan settings profile

haebler, 2024-07-03:
**Describe the bug**
When in the "New Scan" dialog, I can create a new profile for the scanner settings but I am unable to delete an existing profile or make it the default.
 
**To Reproduce**
Steps to reproduce the behavior:
1. Click "Scan"
2. Wait for dialog to open
3. no UI controls to delete a scan configuration profile or make a profile the default

**Expected behavior**
I can remove and modify scan profiles. There seems to be no other place in the UI these are exposed either.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win11
 - Version 7.4.2.0

**Additional context**
NA

cyanfish, 2024-07-03:
Click the "Profiles" button on the toolbar.

closed 2024-07-06 by cyanfish

haebler, 2024-07-07:
What I mean by scan profiles is this. I do not see a way to manage them, other than creating new ones:
[image]

Profiles gets me to the scanner profile, not the scan profile: 
[image]

cyanfish, 2024-07-07:
Those are Windows scanner profiles, not NAPS2. You can try something like this:
https://answers.microsoft.com/en-us/windows/forum/all/editing-windows-fax-scan-scan-profiles/72fac3e7-4bf0-401a-be22-321093ffea1f

Or select "Use predefined settings" instead of "Use native UI" to use NAPS2 profiles.

haebler, 2024-07-07:
@cyanfish Thanks so much for clarifying that!!!! Will set up some NAPS2 profiles and try that.

For all, who are looking for those settings in Windows 11, IMHO the best option is to go through "Windows Scan & Fax"
[image]