cyanfish/naps2 #411: Profile Settings: 400 dpi resolution results in a 300 dpi image, 800 dpi resolution results in a 600 dpi image.

Compunologist, 2024-07-15:
**Describe the bug**
When choosing a resolution of 400 dpi in the Profile Settings a scan will result in a 300 dpi image. When choosing a resolution of 800 dpi a scan will result in a 600 dpi image. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Profiles-->Edit
2. Change Resolution to 400 dpi or 800 dpi
3. Scan the document
4. Save the image
5. Check the image DPI

**Expected behavior**
A chosen resolution of 400 dpi should result in a 400 dpi image, a choice for 800 dpi should result in a 800 dpi image.

**Screenshots**
[image]
_Profile Settings with a 400 dpi scanning resolution selected_

[image]
_The result is an image with a 300 dpi resolution_

The issue is present in both Windows and Linux. I have also tested the available 100, 150, 200, 300, 600 & 1200 dpi settings and they all yield the desired outcome. The incorrect resolution after scanning only occurs at 400 and 800 dpi. NAPS2 does not give an error message, however the debug log mentions "Correcting DPI from 400 to 300":
[debuglog.txt](https://github.com/user-attachments/files/16240019/debuglog.txt)

**Desktop (please complete the following information):**
 - OS: openSUSE Leap 15.6, Windows 10 Home version 22H2, Windows 10 Pro version 22H2
 - Version 7.4.3.0

cyanfish, 2024-07-15:
This happens when the scanner itself doesn't support those DPIs.

Compunologist, 2024-07-15:
The scanner has produced 400 dpi scans without any problems using other software. This is the first time the actual scan falls back to 300 dpi.

cyanfish, 2024-07-15:
In the debuglog you can see on SANE it's definitely not available:
> Int Dpi 4 SoftSelect, SoftDetect WordList 75,100,150,200,300,600,1200

On Windows you could try switching between WIA and TWAIN drivers or reinstalling the manufacturer driver.

Compunologist, 2024-07-15:
Thank you for pointing that out. After some further testing it has become clear it is indeed a driver problem. TWAIN drivers do allow (and honor) the 400dpi setting whereas WIA drivers do not. Would it be possible to hide the unobtainable resolutions in the Profiles Settings? Or maybe present a warning after scanning "requested resolution not supported, DPI corrected to.."? It could save some confusion and misunderstanding.

cyanfish, 2024-07-15:
https://github.com/cyanfish/naps2/issues/183 is the issue tracking that.

closed 2024-07-15 by cyanfish

cyanfish, 2024-08-11:
DPI options now update based on the driver in [7.5.0](https://www.naps2.com/download).

Compunologist, 2024-08-13:
This is a vast improvement. Well done!