cyanfish/naps2 #416: C# - Is there any way to get the scanner name?

Neo6400, 2024-07-19:
Hello

Currently developing an application in C#, I want to know if it is possible to get the scanner name like show in the image below highlighted within the brackets with the software CaptureOnTouch (Canon).

I haven't found anything in the code to retrieve the scanner name.
My scanner has been tested through USB and Ethernet in both case the name doesn't show.

[image]

Thank you

cyanfish, 2024-07-19:
If it's not part of [ScanDevice.Name](https://www.naps2.com/sdk/doc/api/NAPS2.Scan.ScanDevice.html#NAPS2_Scan_ScanDevice_Name) then there's currently no way to get it from NAPS2.

If you want you can try using [NTwain](https://github.com/soukoku/ntwain) (which NAPS2 uses internally) and look at the properties on the [DataSource](https://github.com/soukoku/ntwain/blob/master/src/NTwain/DataSource.cs#L19).

Neo6400, 2024-07-19:
I have tested with NTwain, I checked the datasource but the name of the scanner is nowhere to be found, I have no idea how Canon manage to get the name.

Thank you for you help

closed 2024-07-19 by cyanfish