cyanfish/naps2 #417: Don't make Profiles window modal

trollixx, 2024-07-21:
**Is your feature request related to a problem? Please describe.**

I often need to switch between profiles or adjust profile settings between scans, it's quite annoying that I need to close the window to look at the scan and then open it again.

**Describe the solution you'd like**

Do not set modal flag on the Profiles window.

**Describe alternatives you've considered**

I tried using Profiles toolbar, but there's no way to quickly access profile settings from there. (Could be another enhancement actually, show settings on right click?)

**Additional context**

N/A

Gitoffthelawn, 2024-07-24:
Here's another idea: Allow the `Profiles` window to be docked.

As a useful extension of this idea, perhaps the `Profiles` window and the `Profiles` toolbar could be integrated.  IOW, create a dockable window that shows all your profiles, as well as an `Other profiles` drop-down select button.  If you want, you can include a `Manage` button to allow the user to manage their profiles within this dockable window.

Each profile item within this dockable `Profiles` window will have the same context menu as in the current Profiles window, except with the addition of an `Add to Other` menuitem to add profiles to the `Other profiles` overflow drop-down select menu.

This dockable `Profiles` window will remove the need for the current Profiles toolbar, and eleminate it's deficencies (e.g. inability to edit profile settings).

cyanfish, 2024-08-15:
https://github.com/cyanfish/naps2/issues/337 is my concept for making setting changes easier.

@trollixx Would being able to change paper source/dpi/bit depth/page size from the main window suit your use cases?

trollixx, 2024-08-18:
I think a sidebar should work. It should have something like a combobox for switching between profiles and ability to instantly modify the settings.

cyanfish, 2024-10-05:
I've implemented the sidebar in [8.0b1](https://www.naps2.com/download).

closed 2024-10-05 by cyanfish