cyanfish/naps2 #422: Could not load file or assembly 'Microsoft.Extensions.DependencyInjection.Abstractions, Version=8.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60'. The located assembly's manifest definition does not match the assembly reference. (0x80131040)

PFight, 2024-08-06:
**Describe the bug**
When using NAPS2 in .NET6 application without deps.json getting error "Could not load file or assembly 'Microsoft.Extensions.DependencyInjection.Abstractions, Version=8.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60'. The located assembly's manifest definition does not match the assembly reference. (0x80131040)"

NAPS2 has a target .NET6, but using nuget packages 8.0. If you are try to use NAPS2 in .NET6 application, MSBUILD will produce error:

> error NU1109: Detected package downgrade: Microsoft.Extensions.DependencyInjection.Abstractions from 8.0.1 to centrally defined 6.0.0. Update the centrally managed package version to a higher version.

To solve this error one should add explicit link to it's project to package  Microsoft.Extensions.DependencyInjection.Abstractions  v8.01. Then, after build appliaction will have next strange links:

[image]

All microsoft libraries by default will be 6.0, but these ones, required by NAPS2 will be 8.0. In runtime such strange links has problems in resolving. If you are place deps.json file, then it applies some redirects, and it some how works. But in our app we can't use deps.json, due to dynamic plugin system. **So we can't find a way to use NAPS2 in out .NET6 application at all.**

**To Reproduce**
Steps to reproduce the behavior:
1. Create simple .NET6 application. In project file specify ```<Project Sdk="Microsoft.NET.Sdk.Web">``` and ```<TargetFramework>net6.0</TargetFramework>```
2. Add nuget NAPS2
3. Got error at build "Detected package downgrade"
4. Fix it by adding nuget Microsoft.Extensions.DependencyInjection.Abstractions  v8.01 and others required.
5. Got library version mix of 6.0 and 8.0 in application references.
6. Publish application, remove deps.json, try to run - got error "Could not load file or assembly"

**Expected behavior**
NAPS2 targetting .NET6 should have links to apprepriating 6.0 microsoft nuget packages. Version of NAPS2 for .NET8 should use 8.0 packages.

**Desktop (please complete the following information):**
 - OS: AstraLinux 1.7.5.9 (based on Debian)
 - NAPS2.Sdk Version="1.0.1"

cyanfish, 2024-08-06:
NAPS2.Internals is [LGPL](https://github.com/cyanfish/naps2/blob/master/NAPS2.Internals/LICENSE) so you can modify it according to those terms (i.e. publishing only those changes to NAPS2, not your own software).

Having NAPS2 multi-target packages isn't practical as some of those are from indirect dependencies.

This is what deps.json is for, so aside from modifying the source code yourself, I'm not sure what plugin architecture you're using but trying to accommodate that would be my recommendation.

PFight, 2024-08-07:
Thanks, there is [not actual information in root Readme](https://github.com/cyanfish/naps2?tab=readme-ov-file#license) about License: there is no NAPS2.Internals in the list of LGPL projects . Solution with the fork of naps2 is acceptable.

closed 2024-08-07 by PFight