cyanfish/naps2 #428: On some configurations 7.5.0 broken msi installer

funkerwolf, 2024-08-12:
On some configurations, installing the latest version 7.5.0 does not work properly, installing via GPO does not work. It also looks like it doesn't install properly, there are missing files in program files and no shortcut in the start menu. Also upgrading over from 7.4.3 to 7.5.0 not working.  When trying to install an older version, an error comes up that a more recent version is installed. However, if you uninstall the new version through the installer of the old one, the previous 7.4.3 version is installed without problems.

The x86 version is also missing from the download list on github.

cyanfish, 2024-08-12:
For x86 make sure you click "Show all assets". I'm not sure what the issue you're experiencing is, it works fine for me both as a base installation and when upgrading from 7.4.3. Are you using the exact same installer as you did before? (exe/msi x64/x86)

cyanfish, 2024-08-15:
Closing for now as no one else has reported a similar problem, feel free to comment if there's a reproducible issue.

closed 2024-08-15 by cyanfish