cyanfish/naps2 #433: Save files to the destination folder while scanning

zwo1a, 2024-08-21:
**Is your feature request related to a problem? Please describe.**
When scanning a large number of images, it is often necessary to wait a long time for the save action to complete.

**Describe the solution you'd like**
Provides an option to save image files to a destination folder while scanning to minimize wait time.
When the user makes changes to the image after scanning (after cropping, adjusting colors), the user is prompted to save again.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

cyanfish, 2024-08-21:
Check "Auto Save" in your profile settings.

zwo1a, 2024-08-21:
Sorry, I missed that option. This issue has been resolved.

closed 2024-08-21 by zwo1a