cyanfish/naps2 #440: [Feature request] Automatic flagging and deletion of blank pages

zwo1a, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
When scanning mixed single and double-sided documents, they can contain many blank pages. We need to delete them manually.

**Describe the solution you'd like**
1. The program scans automatically determines if a page is blank and excludes it.
2. The program flags blank pages and leaves it up to the user to determine whether or not to delete them.
3. The program saves blank pages to a separate folder so that the user can subsequently recover pages that may be useful. (Add some symbols and numbers after the serial number to allow for sorting using the file name, e.g. a blank page following 100 pages is labeled 100-blank01.)

**Describe alternatives you've considered**
Delete them manually in NAPS2;
Save first, then sort by file size to identify possible blank pages and delete them manually.

**Additional context**
N/A

cyanfish, 2024-08-26:
In your profile settings, under Advanced, check "Exclude blank pages".

zwo1a, 2024-08-27:
> In your profile settings, under Advanced, check "Exclude blank pages".

Thank you very much for your reply. I have looked into this feature but I am still having some difficulty in some cases.
For example, some of the scanner outputs are not very clean, as well as scanning non-white paper pages. And the program will ignore these images at that point without any prompting.
I think that adding the following features would help (still the same 2 and 3 mentioned above):

1. display “xxx blank pages excluded” somewhere in the UI.
2. by clicking this button the user can browse the blank pages identified by the program and confirm whether to restore them or save them to a separate folder.
3. or save them in a separate folder.