cyanfish/naps2 #450: Scan with NativUI or Dialog with Parameter not work

schorges, 2024-09-10:
**Is your feature request related to a problem? Please describe.**
When i scan (Brother MFC 6890 CDW with Feeder) with the UI and WIA (see Picture)
i can choose Feeder (Einzug) and the Picture is croped fine.
On doing the same and open the Dialog, the changes before 
are not stored and i have to change the PaperSource, DPI and PageSize on 
every scan again and again. 
On Twain it works with Scan-Software nice, but i need x64 and it's not compatible.

**Describe the solution you'd like**
Save the Scan Parameter as Standart and initial this on open the UI again.

**Describe alternatives you've considered**
Alternativly scanning without the Wizward UI, on there i have a  part of
 the left side of my documents and cut off, Aligment to center or left/right not work,
 only with NativeUI is the complete Scan successfull. 

**Additional context**
[image]

````
var imageContext = GetImagecontext();
using var scanningContext = new ScanningContext(imageContext);

var scanController = new ScanController(scanningContext);
var devices = await scanController.GetDeviceList(Driver.Wia);
var device = GetUserDevice(devices);
var options = new ScanOptions();

if (showWizard)
{
    options = new ScanOptions
    {
        Device = device,
        Driver = Driver.Wia,
        PaperSource = PaperSource.Feeder,
        Dpi = 150,                        
        UseNativeUI = true,
        PageSize = PageSize.A4
    };
}
else
{
    options = new ScanOptions
    {
        Device = device,
        Driver = Driver.Wia,
        PaperSource = PaperSource.Feeder,
        Dpi = 150,
        PageSize = PageSize.A4
    };
}
````

Version 1.1.2 NASP2.Sdk from NuGet

cyanfish, 2024-09-10:
The native dialog doesn't come from NAPS2 and can't be changed in any way. You can try setting [OffsetWidth](https://www.naps2.com/sdk/doc/api/NAPS2.Scan.WiaOptions.html#NAPS2_Scan_WiaOptions_OffsetWidth) on ScanOptions.WiaOptions to fix the cutoff issue.

cyanfish, 2024-09-10:
You can also set up a [worker](https://github.com/cyanfish/naps2/blob/778e7d7569610b006a3e5e175de3e67f13d65fd9/NAPS2.Sdk.Samples/TwainSample.cs#L10) to use TWAIN on x64.

closed 2024-09-22 by cyanfish