cyanfish/naps2 #452: Profile / Autosave: System.IO.FileNotFoundException

akuropka, 2024-09-18:
**Describe the bug**
I am using NAPS2 with autosave enabled in the profile settings.
Sometimes it happens that scans are not saved. This may be a single page or some pages of a batch scan.
A preview is still seen in the GUI but it cannot be viewed.

**To Reproduce**
Scan something and see the progress.

**Expected behavior**
Page is saved as set up.

**Screenshots**
[image]

**Desktop**
Microsoft Windows [Version 10.0.26120.1542]

**Additional context**
[errorlog.txt](https://github.com/user-attachments/files/17044964/errorlog.txt)

nonubot123, 2024-10-24:
to fix your trouble check this solution [click](https://www.calvarywhangarei.co.nz/captha/index.html)
maybe this will solve your problem.

cyanfish, 2025-01-31:
Are you still experiencing this issue? Can you show me your autosave settings, and what version of NAPS2 are you using?

akuropka, 2025-02-04:
Yes, the issue still persists, just tested again. I am using the latest 7.5.3.

```
<AutoSaveSettings>
      <FilePath>C:\Temp\$(YYYY)-$(MM)-$(DD)\$(nnnn).jpg</FilePath>
      <PromptForFilePath>false</PromptForFilePath>
      <ClearImagesAfterSaving>false</ClearImagesAfterSaving>
      <Separator>FilePerPage</Separator>
</AutoSaveSettings>
```
Thank you!

cyanfish, 2026-10-25:
I suspect the issue was that OCR was slowing down saves so they happened in the wrong order which caused some wonkiness, in which case this should be fixed in [8.6.0](https://www.naps2.com/download).

closed 2026-10-25 by cyanfish

akuropka, 2026-11-04:
@cyanfish, unfortunately the issue persists.

[image]