cyanfish/naps2 #455: No OCR happening after importing PDF

neo269, 2024-09-21:
**Describe the bug**
No OCR happening after importing Document

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Import'
2. Click on 'PDF'
3. PDF gets imported
4. Go to 'Save PDF' 
5. After Saving PDF When opened, dont see OCR'd PDF. Its Saves same PDF as imported

OCR Language is Gujarati

**Expected behavior**
Expected an OCRd PDF

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 7.5.1

**Additional context**
Trying since two days but not able to get OCR working on my pdf. Attaching just one page if someone can help me out
[test.pdf](https://github.com/user-attachments/files/17084595/test.pdf)
[image]

cyanfish, 2024-09-21:
NAPS2 won't do OCR on a page that already has text (in this case the header/footer).

neo269, 2024-09-21:
> NAPS2 won't do OCR on a page that already has text (in this case the header/footer).

So if I crop the pdf of header & footer, will OCR work ?

cyanfish, 2024-09-21:
Well, technically any change you do in NAPS2 (even just cropping 1 pixel) will rasterize the PDF (turning it fully into an image without any text or anything) which will let OCR work, though it might change the PDF quality/file size. Or if you had a PDF editor that could just remove the text that would work too.

neo269, 2024-09-21:
> Well, technically any change you do in NAPS2 (even just cropping 1 pixel) will rasterize the PDF (turning it fully into an image without any text or anything) which will let OCR work, though it [...]

Thanks for the insight.
Apparently I just tried to OCR an **English** Language PDF from the net & it did work
However, when I tried your suggestion with the cropped PDF for **Gujarati** Language, it did not work. (cropped used Acrobat)

Attaching both PDFs if you need to test.

[english_sample.pdf](https://github.com/user-attachments/files/17085034/english_sample.pdf)
[gujarati_cropped_sample.pdf](https://github.com/user-attachments/files/17085035/gujarati_cropped_sample.pdf)

cyanfish, 2024-09-21:
Whatever application you used to crop the header/footer left the text there, just off page - it would need to be fully deleted. But yes, converting to an image should work (or doing the crop in NAPS2 etc).

neo269, 2024-09-22:
> Whatever application you used to crop the header/footer left the text there, just off page - it would need to be fully deleted. But yes, converting to an image should work (or doing the crop in NAPS2 etc).

Thanks for all the insights. Especially your first comment!

closed 2024-09-22 by cyanfish