cyanfish/naps2 #459: naps 7.x: OCR not executed on imported PDFs (works with 6.x)

n00ris, 2024-09-24:
**Describe the bug**
OCR settings are enabled. 
After I import a pdf that does not have an ocr layer yet and save it again to a new pdf file, the new file should contain OCR text.

https://www.naps2.com/doc/ocr

It works with naps2 Version 6.1.2

It is easy to reproduce with the protable versions.

Last GOOD version
naps2-6.1.2-portable OK

First BAD Version
naps2-7.0b1-win BROKEN

Latest version (naps2-7.5.1-win) is still broken.

**To Reproduce**
Steps to reproduce the behavior:
1. Download portable version as metioned above and extract. Open NAPS2.Portable.exe.
2. In the UI, go to 'OCR'
3. Download German language (I only checked "German", I un-checked "English")
4. Check "Make PDFs searchable using OCR"
5. Choose OCR language "German"
6. Choose OCR mode "Best"
7. Check "Automatically run OCR after scanning"
8. Click on 'OK'
9. Click on 'Import'
10. Choose a pdf with german content and no OCR yet
11. Click "Save PDF" to save everything again as a new pdf.
12. Open the PDF in a pdf viewer. With version 6.1.2 you will have OCR text. With 7.0b1 and above you have no OCR text.

**Expected behavior**
OCR text layer should be present

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10

**Additional context**
-

cyanfish, 2024-09-24:
Can you attach a PDF with the issue? Also see https://github.com/cyanfish/naps2/issues/455

n00ris, 2024-09-25:
Hello, from the documentation, I was aware of this line "NAPS2 follows a simple rule: if a page already has text, it leaves it alone; if it doesn't, OCR will be used."

So before I filed this issue, I tested the following: 
- Opened the original pdf
- Run ctrl-a and in windows editor ctrl-v -> no characters inserted

And I read print to pdf feature should remove OCR, so I did:
- I opened the original pdf
- used microsoft print to pdf
- importet this version
- same result (no ocr)
[image]

cyanfish, 2024-09-25:
Well, it works for me in general, so I can't do anything about it without a sample PDF that has the problem.

n00ris, 2024-09-25:
Unfortunately the pdf contains some sensitive data. So I imported the first page, and saved it with naps2 as jpeg image. Then I imported the image again and saved it as pdf. Same result. But I may found something, I do not have an extra pdf viewer on this system except edge. Let me test something

n00ris, 2024-09-25:
If the pdf compatibility is "Standard", which version is produced? And did it change between 6.1.2 and 7 beta?

[image]

cyanfish, 2024-09-25:
Edge might not be allowing you to select text, see https://sourceforge.net/p/naps2/discussion/general/thread/4d2f0aeb42/?limit=25#a112

PDF version should be 1.4 and hasn't changed

n00ris, 2024-09-25:
> PDF version should be 1.4 and hasn't changed

I mean one of the list: PDF/A-1b to 3u?

cyanfish, 2024-09-25:
A2/A3 are 1.7 instead of 1.4

n00ris, 2024-09-25:
Ok you are right it is the edge issue after all!
I was wondering since I could select text in the print preview screen!
The steps from the issue you posted are working.

I'm still wondering why the ocr in the pdf created with naps2 6.1.2 is working in edge without those changes.
But that is not part of this issue. Close this myself.

closed 2024-09-25 by n00ris