cyanfish/naps2 #463: Strange highlight when searching OCR'ed PDF

softfoot, 2024-10-03:
**Describe the bug**
I often import PDFs and save them to get a searchable PDF, usually it works very well and when I search
for a phrase it highights it nicely.
However, with some PDFs instead of highlighting the phrase it shows a verical bar, see picture.
It is not the pdf reader (Foxit) as I see the same result if I open the file in Firefox and Chrome (unless they all use the same OCR engine!!)

**To Reproduce**
Steps to reproduce the behavior:
1. Import test PDF
2. Make sure OCR is enabled
3. Save as new PDF
4. Open new PDF in viewer
5. Search for "GPIB address"
6. View problem

[image]

**Expected behavior**
The phrase should be highlighted in the PDF viewer.

**Desktop (please complete the following information):**
Windows 7
NAPS Version 7.5.1.0

**Additional context**
If you require a PDF that shows this problem I can upload it, but it's big 3.2Mb

nonubot123, 2024-10-24:
to fix your trouble check this solution [click](https://www.calvarywhangarei.co.nz/captha/index.html)
maybe this will solve your problem.

cyanfish, 2026-10-12:
Please try with [8.5.0](https://www.naps2.com/download) and see if you still have the issue.

closed 2026-10-12 by cyanfish