cyanfish/naps2 #464: Import PDF Files

UstimVV, 2024-10-03:
**Describe the bug**
When I try to import PDF into NAPS2 7.5.1 I ha

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Import'
2. Choose any pdf file
3. Click on 'Open'
4. See error: The file 'filename.pdf' could not be imported...
5. Technical details: System.Exception: Could not load library: "libpdfium.so". Error:
see on screenshot
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop:**
 - OS: CentOS
 - Version 7.5.1

cyanfish, 2024-10-03:
What version of CentOS?

UstimVV, 2024-10-04:
kernel version 6.6.51
centos 8

sT331h0rs3, 2024-10-04:
I had such issue, when `naps2` executable was incorrectly placed in `/usr/bin` directory instead of `/usr/lib/naps2`.
The error occurs because `libpdfium.so` is lazy-loaded from path `./_linux/libpdfium.so`, relative to the `naps2` executable. The full path is `/usr/lib/naps2/_linux/libpdfium.so` by default.

UstimVV, 2024-10-04:
Executable naps2 I found in /usr/bin. I made symlink /usr/lib/naps2 -> /usr/lib64/naps2/ where I found file _linux/libpdfium.so. But I have the same error.

sT331h0rs3, 2024-10-04:
Executable `naps2` should be placed in `/usr/lib/naps2` directory by default (could be `/usr/lib64/naps2` in some distros as well).
File `/usr/bin/naps2` might be a symlink to `/usr/lib/naps2/naps2` though.

cyanfish, 2024-10-04:
The other possibility is that Centos is using too old a glibc (2.28) for the pdfium build (it's supposed to only require 2.27 but I might have missed some change). I'll check later but if you want you can see what glibc version libpdfium.so depends on.
https://stackoverflow.com/questions/3436008/how-to-determine-version-of-glibc-glibcxx-binary-will-depend-on

UstimVV, 2024-10-04:
1. Executable naps2 I put to diffrents places - no success.
2. glibc version now 2.36

cyanfish, 2024-10-04:
What's the output of `ldd /lib/naps2/_linux/libpdfium.so`?

UstimVV, 2024-10-07:
ldd /lib/naps2/_linux/libpdfium.so
ldd: warning: you do not have execute permissions for `/lib/naps2/_linux/libpdfium.so'
is not a dynamic executable

UstimVV, 2024-10-07:
then I use naps2-7.1.2 version output is:
ldd libpdfium.so
	linux-vdso.so.1 (0x00007fffe869d000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f7cb17c4000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f7cb16e4000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f7cb11e0000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f7cb1003000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f7cb17df000)

cyanfish, 2024-10-07:
That's... odd. It works fine on my Fedora VM (which also uses the RPM). At some point I might try creating a CentOS VM and see if I can track it down but I'm not sure what the problem is.

UstimVV, 2024-10-07:
Then I use version naps2-7.1.2 I can import pdf files and output is:
# ldd /lib/naps2/_linux/libpdfium.so 
	linux-vdso.so.1 (0x00007ffe9c199000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fbe15ba5000)
	libm.so.6 => /lib64/libm.so.6 (0x00007fbe15520000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fbe15b85000)
	libc.so.6 => /lib64/libc.so.6 (0x00007fbe15343000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fbe15bc0000)
and I can im
After update to version naps2-7.5.1 output is:
# ldd /lib/naps2/_linux/libpdfium.so 
ldd: warning: you do not have execute permissions `/lib/naps2/_linux/libpdfium.so'
	linux-vdso.so.1 (0x00007ffea496a000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fea86bed000)
	libm.so.6 => /lib64/libm.so.6 (0x00007fea86b0d000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fea86aed000)
	libc.so.6 => /lib64/libc.so.6 (0x00007fea86423000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fea86c08000)

UstimVV, 2024-10-07:
> That's... odd. It works fine on my Fedora VM (which also uses the RPM). At some point I might try creating a CentOS VM and see if I can track it down but I'm not sure what the problem is.

Thank you

nonubot123, 2024-10-24:
to fix your trouble check this solution [click](https://www.calvarywhangarei.co.nz/captha/index.html)
maybe this will solve your problem.

UstimVV, 2024-10-25:
Import pdf files on Version 7.5.1-2 working exelent)

closed 2025-01-02 by cyanfish