cyanfish/naps2 #469: PDF by Email - kubuntu

sulicz, 2024-10-11:
When I want to send PDF by email, I click button and get error from Thunderbird "File can not be attached  /home/user/.config/naps2/temp/ruc3fiy5.0hn/Scan.pdf,  because it not exist". (Sorry of translate - I have it in Czech).

The file do not exist in filesystem, even the directory ruc3fiy5.0hn is not there. I checked permissions on dirs and tried to delete temp directory. It create new bud subdirecory and scan.pdf no. 

**Desktop (please complete the following information):**
 - OS: Kubuntu 
 - Version 24.0

nonubot123, 2024-10-24:
to fix your trouble check this solution [click](https://www.calvarywhangarei.co.nz/captha/index.html)
maybe this will solve your problem.

cyanfish, 2024-12-30:
Are you using the flatpak?

sulicz, 2025-01-06:
I use native package from ubuntu repo

sulicz, 2025-07-08:
> I got the same issue if Thunderbird has already been started. If Thunderbird is not running the function does work for me.

I can confirm this behavior.  But I do not understand why is it like that.

MadTooler, 2025-12-28:
Confirming this is still an open issue and with behavior as noted prior.

I also watched the parent folder /home/user/.config/naps2/temp with Thunderbird open and saw the temp pdf file folder generated but removed before the draft email fully opened.  Maybe a timing or trigger issue?

Other than this, naps2 is my favorite linux based scan utility.  I appreciate the layout and approach.  Thanks!

gianlucaromito, 2026-04-28:
I have the same behavior as described above. Thunderbird closed, no problem, Thundebird open file not found error.
Linux Mint 22.3.
Thanks.
ps. naps2 is the best scanning tool on linux... Great job.

cyanfish, 2026-10-25:
This should be fixed as of 8.4.0.

closed 2026-10-25 by cyanfish