cyanfish/naps2 #473: Placeholders in PDF Settings "Default File Path" do not work

Gitoffthelawn, 2024-10-16:
**Describe the bug**
The placeholders in the PDF Settings dialog box's "Default File Path" input field do not work.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the PDF Settings dialog box
2. Enter a valid path, such as `C:\NAPSbug\Bug Test\NAPS Test $(YYYY)-$(MM)-$(DD) $(hh)_$(mm)_$(ss).pdf`
3. Click on the "Placeholders" link.
4. Notice that the preview is correct.
5. Perform a scan.
6. Click the "Save PDF" toolbar button to save the PDF.
7. When the dialog box opens to save the PDF, the filename will be `NAPS Test $(YYYY)-$(MM)-$(DD) $(hh)_$(mm)_$(ss).pdf` (in this example).

**Expected behavior**
Placeholders to work.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 7.5.2.0

cyanfish, 2024-10-16:
If you actually proceed to save it should work. Replacement happens after.

I understand that can be a bit confusing, but it kind of has to be that way, at least for $(nnn), as if you have Single Page PDFs checked it will create multiple file names with different numbers. (I guess replacing dates only could work but at the moment all placeholders are done together.)

nonubot123, 2024-10-24:
to fix your trouble check this solution [click](https://www.calvarywhangarei.co.nz/captha/index.html)
maybe this will solve your problem.

cyanfish, 2025-03-29:
This should be fixed in [8.1.2](https://www.naps2.com/download).

closed 2025-03-29 by cyanfish