cyanfish/naps2 #474: Get WIA data stream using the SDK

pingarelho, 2024-10-16:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
I'd like a way to get the WIA data stream as it is being written instead of getting just the finished image (using the SDK).

**Describe alternatives you've considered**
I've considered looking into WIA examples and creating my own C# app that does this, but it would be way easier/faster to use NAPS2 profiles and getting the data that way.

**Additional context**
I'm trying to turn an old scanner into a musical instrument that generates sound as it is scanning an image.

nonubot123, 2024-10-24:
to fix your trouble check this solution [click](https://www.calvarywhangarei.co.nz/captha/index.html)
maybe this will solve your problem.

cyanfish, 2025-01-26:
It's technically possible to expose the [underlying stream](https://github.com/cyanfish/naps2-wia/blob/e71758f79089192b1ecaacab37f578a5968ad302/NAPS2.Wia/Native/WiaTransferCallback2.cs#L28) as part of progress reporting but that's fairly complicated and low priority right now. Hypothetically this could support a visual display of the scan in progress in the NAPS2 app as some other apps do.