cyanfish/naps2 #475: Imprint capabilities

RaHorusFreak, 2024-10-17:
Hello team! I am migrating from NTwain to NAPS2.Sdk, and I'm having difficulty trying to use the imprinter option. With native Twain libraries, you can imprint directly on the sheet you're scanning (without the need to do it later when you already have the image in memory):
```c#
if(dataSource.Capabilities.CapPrinterEnabled.CanSet)
    dataSource.Capabilities.CapPrinterEnabled.SetValue(BoolType.True);
if(dataSource.Capabilities.CapPrinter.CanSet)
    dataSource.Capabilities.CapPrinter.SetValue(Printer.ImprinterBottomAfter);
if(dataSource.Capabilities.CapPrinterMode.CanSet)
    dataSource.Capabilities.CapPrinterMode.SetValue(PrinterMode.SingleString);
if(dataSource.Capabilities.CapPrinterCharRotation.CanSet)
    dataSource.Capabilities.CapPrinterCharRotation.SetValue(-90);
if(dataSource.Capabilities.CapPrinterString.CanSet)
    dataSource.Capabilities.CapPrinterString.SetValue(stampString);
``` 
Similarly, with NTwain I can find out if the device supports this action:
```c#
dataSource.Capabilities.CapPrinterEnabled.CanSet && dataSource.Capabilities.CapPrinterString.CanSet
``` 
Is there any workaround to archieve this using NAPS2?
Thanks in advance.

nonubot123, 2024-10-24:
to fix your trouble check this solution [click](https://www.calvarywhangarei.co.nz/captha/index.html)
maybe this will solve your problem.

RaHorusFreak, 2024-11-20:
Hello, it´s this repo dead?